ros2/ros2cli #233: ros2 lifecycle get always shows state of all nodes even when selecting one by name

ralph-lange, 2019-05-08:
## Bug report

When calling ros2 lifecycle get /nodename, the state of _all_ nodes is shown. Expected behavior is that only the state of the selected node is shown.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Crystal from binary, Pre-Dashing from source
- Version or commit hash:
  - 1766de05961a7a08a77f446a22fca0abdd6d0552
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
[...]:~/ros2_dashing_ws$ ros2 lifecycle get
/damped_pendulum/damped_pendulum: inactive [2]
/transport_delay/transport_delay: unconfigured [1]
[...]:~/ros2_dashing_ws$ ros2 lifecycle get /damped_pendulum/damped_pendulum 
inactive [2]
unconfigured [1]
[...]:~/ros2_dashing_ws$ 
```

#### Expected behavior
Only the state of the selected node is shown.

#### Actual behavior
States of all nodes are shown - without names.

ralph-lange, 2019-05-08:
@Karsten1987 See https://github.com/ros2/ros2cli/issues/233 for the issue with the two (or more) output lines with `ros2 lifecycle get /nodename`

dirk-thomas, 2019-05-08:
Currently also all states are being queried even if only a specific node is requested: https://github.com/ros2/ros2cli/blob/1766de05961a7a08a77f446a22fca0abdd6d0552/ros2lifecycle/ros2lifecycle/verb/get.py#L52

PR #266 (2019-05-29): [ros2lifecycle] Only return the state for the node requested
Previously, the 'get' verb was returning the state for all lifecycle nodes.

Fixes #233
closed 2019-05-29 by jacobperron