ros2/ros2cli #235: [ros2topic] 'The passed message type is invalid'

jacobperron, 2019-05-14:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 (and presumably all other platforms)
- Installation type:
  - source
- Version or commit hash:
  - tag 0.7.2 (8f419664cd37a42702365c72e88cdbf28ee7b18c)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  -  rclpy

#### Steps to reproduce issue

Start publishing to a topic:
```
ros2 topic pub /chatter std_msgs/Empty
```

Try using the verbs `bw`, `hz`, and `delay`. E.g.
```
ros2 topic bw /chatter
```

#### Expected behavior

The tools work (e.g. echo bandwidth for `bw`).

#### Actual behavior

All tools log the message **"The passed message type is invalid"** and exit.

#### Additional information

It's likely this is a bug introduced by https://github.com/ros2/ros2/issues/677.

cottsay, 2019-05-15:
I'm using this slightly hacky change to work around this issue for now:
```diff
--- a/ros2topic/ros2topic/api/__init__.py
+++ b/ros2topic/ros2topic/api/__init__.py
@@ -57,9 +57,11 @@ class TopicNameCompleter:
 def import_message_type(topic_name, message_type):
     # TODO(dirk-thomas) this logic should come from a rosidl related package
     try:
-        package_name, message_name = message_type.split('/', 2)
+        package_name, message_name = message_type.split('/', 1)
         if not package_name or not message_name:
             raise ValueError()
+        if message_name.startswith('msg/'):
+            message_name = message_name[4:]
     except ValueError:
         raise RuntimeError('The passed message type is invalid')
```

I think the `maxsplit` value should always have been 1, even before the regression.

jacobperron, 2019-05-15:
Thanks. Yeah, I think `maxplit` should have been one.

PR #237 (2019-05-15): [ros2topic] Handle multiple namespace parts in message type
Fixes #235.
Now the 'bw', 'hz', and 'delay' verbs work again.
commit 52b5a70f (2019-05-15): [ros2topic] Handle multiple namespace parts in message type (#237)

Fixes #235.
Now the 'bw', 'hz', and 'delay' verbs work again.
closed 2019-05-15 by jacobperron