ros2/ros2cli #248: ros2 service list --spin-time does spin for the given duration

jacobperron, 2019-05-22:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Debian packages
- Version or commit hash:
  - 0.7.3
- DDS implementation:
  - Connext
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
ros2 service list --spin-time 10
```

#### Expected behavior

The command waits for 10 seconds (as documented) until returning with a list of services.

#### Actual behavior

The command returns immediately.

#### Additional information

The behavior doesn't change if there are services to list or not.

dirk-thomas, 2019-05-22:
The behavior is actually expected if you already have a daemon process running. The spin time only applies if the invocation doesn't use the daemon and has to create its own now. See #253 to clarify this in the help message.

closed 2019-05-22 by dirk-thomas

shmpwk, 2023-08-07:
@dirk-thomas 
Do you mean when using `--spint-time`, we should set `--no-daemon` ?
https://github.com/ros2/ros2cli/blob/351ef3c7442f49013d84084dea23fe399517690f/ros2cli/ros2cli/node/strategy.py#L77-L79

fujitatomoya, 2023-08-07:
> Do you mean when using --spint-time, we should set --no-daemon ?

if you do not want to use daemon, that is correct. instead of setting `--spin-time`, 0.5 sec in default.

precisely, `--spin-time` is effective either `--no-daemon` is specified or only when daemon is not running. (daemon will be spawned via ros2cli command, if that is not running.)

fujitatomoya, 2024-03-16:
> ros2 topic echo --spin-time 5 --no-daemon /cmd_vel_or_whatever_not_published still failed to hold the expected spin-time

i think this is expected behavior. just looking for the topic to spin without daemon, and after given period, it fails.

> Surprisingly, ros2 topic echo --spin-time 0.5 would exit just as expected, and report

i think daemon is running, so it can look up to the cached graph data, and then fails immediately because there is no such topic published yet.

> help wanted, thank you. Ubuntu 22.04, with latest ros-humble-desktop.

i do not see any problems here, can you elaborate what the problem here? probably creating another issue would be better since this one is closed and old, and technically this is the issue for `list` verb.

PR #253 (2019-05-22): update help of --spin-time
Fixes #248.