ros2/ros2cli #260: [ros2component] standalone verb creates zombie containers

jacobperron, 2019-05-29:
## Bug report

**Required Info:**

- Operating System:
  - macOS 
- Installation type:
  - Binaries 
- Version or commit hash:
  - 
- DDS implementation:
  - Fast-RTPS 
- Client library (if applicable):
  - N/A 

#### Steps to reproduce issue

Run the `ros2 component standalone` with invalid arguments, for example:
```
ros2 component standalone foo bar
```

#### Expected behavior

The command fails and no component container is running.

#### Actual behavior

The command fails with the output:

```
$ ros2 component standalone foo bar
[ERROR] [standalone_container_d28808be9256]: Could not find requested resource in ament index
Failed to load component: Could not find requested resource in ament index
```

But, we can see a container left running:

```
$ ros2 component list
/standalone_container_d28808be9256
```

Multiple invalid calls will result in multiple zombie containers.

mjcarroll, 2019-05-29:
I can confirm this behavior on `Bionic` source build.

closed 2019-05-29 by mjcarroll

jubeira, 2019-05-30:
This doesn't seem to happen in Dashing testing binaries for Linux :+1:

PR #269 (2019-05-29): [ros2component] Stop the standalone container when load fails
This prevents zombie container processes from hanging around when a
component fails to load.

Closes: #260