ros2/ros2cli #262: [ros2msg] Inconsistent message naming in the different verbs

ivanpauno, 2019-05-29:
**Required Info:**

- Operating System:
  - Windows server 2019 datacenter
- Installation type:
  - from source
- Version or commit hash:
  - current master
- DDS implementation:
  - FastRTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
1- `ros2 msg show std_msgs/Int64`
2- `ros2 msg show std_msgs/msg/Int64`
3- `ros2 msg package std_msgs`
4- `ros2 msg list`

#### Expected behavior

Option 1 above shouldn't work. 3 and 4 should show the messages in the `action_msgs/msg/GoalInfo` format, and not like `action_msgs/GoalInfo`. Option 2 should work.

#### Actual behavior

Option 1 works, when it shouldn't. Option 2 also works, same output as 1.

3-
```
std_msgs/Bool
std_msgs/Byte
std_msgs/ByteMultiArray
std_msgs/Char
...
```
I would expect the output to be `std_msgs/msg/Bool`, etc.

4- It works like expected:
```
action_msgs/msg/GoalInfo
action_msgs/msg/GoalStatus
action_msgs/msg/GoalStatusArray
...
```

#### Additional information

Haven't reproduced it in other platforms or with other DDS vendors, but I suppose the same is happening.

clalancette, 2019-05-29:
> Option 1 above shouldn't work.

I don't agree with this, mostly for backwards compatibility reasons.  That worked in both ROS 1 and in previous versions of ROS 2, so I think it should continue to work for continuity.  And it is clear that you are asking about a message (not a service or action), so there is no ambiguity there.

closed 2019-05-29 by dirk-thomas

jubeira, 2019-05-30:
Just for the record, this is fixed in Ubuntu 18 Dashing testing binaries :+1:

PR #263 (2019-05-29): fix message/service naming in package verbs
Fixes #262.