ros2/ros2cli #276: ros2 topic info to show topic types

sloretz, 2019-06-07:
## Feature request

#### Feature description
It would be nice if `ros2 topic info` showed the types on the topic. Currently it just shows the count of publishers and subscribers.

```
$ ros2 topic info /amcl_pose
Topic: /amcl_pose
Publisher count: 0
Subscriber count: 1
```

As a workaround, I can use `ros2 topic list -t`. That shows all topics, but it would be nice to have on `info` because sometimes I only care about one.

```
$ ros2 topic list -t
# ...
/amcl_pose [geometry_msgs/msg/PoseWithCovarianceStamped]
# ...
```

#### Implementation considerations

Here is where `ros2 topic list` gets the type. This would need to be copied

https://github.com/ros2/ros2cli/blob/9e1cb8d645b1956ef29588a4332f6ec660874bd2/ros2topic/ros2topic/verb/list.py#L38-L41

dirk-thomas, 2019-06-07:
Isn't this what #204 is implementing?

sloretz, 2019-06-07:
> Isn't this what #204 is implementing?

Gah, I failed at searching

closed 2019-06-07 by sloretz