ros2/ros2cli #277: ros2 topic echo does not support messages generated directly from IDL

davidhodo, 2019-06-08:
## Bug report
- Operating System:
  -MacOS 10.14.5 (Mojave)
- Installation type:
  - Binaries
- Version or commit hash:
  - Dashing
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
ROS2 topic echo currently hardcodes '.msg' for importing message python modules.  Message packages created directly from idl files as supported in Dashing use a '.idl' name.  Running 'ros2 topic echo' on a topic that uses a message generated from idl fails.

An example package that creates a simple string message using idl can be found at: [ros2_idl_demo](https://github.com/davidhodo/ros2_idl_demo).  Building the message (idl_msgs) and sample publisher (idl_pub) packages and then running `ros2 run idl_pub idl_pub` and `ros2 topic echo /chatter` results in a crash.

#### Expected behavior
```
---
var_string: 'ROS2: Hello'
---
var_string: 'ROS2: Hello'

```

#### Actual behavior

```
davidhodo@Davids-MacBook-Pro:idl_test_ws∫ ros2 topic echo /chatter
Traceback (most recent call last):
  File "/opt/ros/dashing/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.7.0', 'console_scripts', 'ros2')()
  File "/opt/ros/dashing/lib/python3.7/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/dashing/lib/python3.7/site-packages/ros2topic/command/topic.py", line 43, in main
    return extension.main(args=args)
  File "/opt/ros/dashing/lib/python3.7/site-packages/ros2topic/verb/echo.py", line 69, in main
    return main(args)
  File "/opt/ros/dashing/lib/python3.7/site-packages/ros2topic/verb/echo.py", line 79, in main
    node.node, args.topic_name, args.message_type, callback)
  File "/opt/ros/dashing/lib/python3.7/site-packages/ros2topic/verb/echo.py", line 106, in subscriber
    msg_module = import_message_type(topic_name, message_type)
  File "/opt/ros/dashing/lib/python3.7/site-packages/ros2topic/api/__init__.py", line 71, in import_message_type
    module = importlib.import_module(package_name + '.' + middle_module)
  File "/usr/local/Cellar/python/3.7.3/Frameworks/Python.framework/Versions/3.7/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'idl_msgs.msg'
```

#### Additional information

The error comes from the import_message_type method in  ros2topic/ros2topic/api/__init__.py.  It currently hardcodes a middle_module name of '.msg'.  For message packages created from idl this should be '.idl'.

dirk-thomas, 2019-06-08:
Thanks for reporting this. #223 described s new command which should be able to handle all kinds including `.idl` based interfaces.

clalancette, 2026-11-14:
This was fixed in #415 quite a while ago, so closing.

closed 2026-11-14 by clalancette