ros2/ros2cli #286: Intermittent failure to list node info

rotu, 2019-06-19:
Sometimes calling ros2 node info for an active node does not show any info in `ros2 node info`. This is seemingly random and doesn't show any diagnostic info.

This seems to only happen when the introspected node is running under FastRTPS, though it has been observed even when running the ros2 node tool under OpenSplice.

$ for i in {1..10}; do ros2 node info /talker && sleep 1; done
```
/talker
  Subscribers:

  Publishers:

  Services:

/talker
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /chatter: std_msgs/msg/String
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Services:
    /talker/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /talker/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /talker/get_parameters: rcl_interfaces/srv/GetParameters
    /talker/list_parameters: rcl_interfaces/srv/ListParameters
    /talker/set_parameters: rcl_interfaces/srv/SetParameters
    /talker/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
/talker
  Subscribers:

  Publishers:

  Services:

/talker
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /chatter: std_msgs/msg/String
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Services:
    /talker/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /talker/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /talker/get_parameters: rcl_interfaces/srv/GetParameters
    /talker/list_parameters: rcl_interfaces/srv/ListParameters
    /talker/set_parameters: rcl_interfaces/srv/SetParameters
    /talker/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
/talker
  Subscribers:

  Publishers:

  Services:

/talker
  Subscribers:

  Publishers:

  Services:

/talker
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /chatter: std_msgs/msg/String
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Services:
    /talker/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /talker/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /talker/get_parameters: rcl_interfaces/srv/GetParameters
    /talker/list_parameters: rcl_interfaces/srv/ListParameters
    /talker/set_parameters: rcl_interfaces/srv/SetParameters
    /talker/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
/talker
  Subscribers:

  Publishers:

  Services:

/talker
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /chatter: std_msgs/msg/String
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Services:
    /talker/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /talker/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /talker/get_parameters: rcl_interfaces/srv/GetParameters
    /talker/list_parameters: rcl_interfaces/srv/ListParameters
    /talker/set_parameters: rcl_interfaces/srv/SetParameters
    /talker/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
/talker
  Subscribers:

  Publishers:

  Services:

```

jacobperron, 2019-06-25:
Please fill out the required info in the issue template.

What OS?
Which version of ROS are you using?
Is the installation type binary or from source? 
Are you running one of the demo nodes? If so, which one?

rotu, 2019-06-25:
This is Ubuntu Linux 18.04.2
ROS built from the master source.
The running node is `ros2 run demo_nodes_cpp talker`.

It seems that this has since been resolved - now occasionally `ros2 node info talker` fails with the following message, which is *much* better than silent failure:

```
[ERROR] [rmw_fastrtps_shared_cpp]: Unable to find GUID for node: talker
Failed to get_subscriber_names_and_types: Unable to find GUID for node , at /opt/ros/master/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_node_info_and_types.cpp:85
/talker
```

closed 2019-06-25 by rotu