ros2/ros2cli #309: Interest in rosshow in ros2cli?

SteveMacenski, 2019-07-31:
rosshow is a CLI for displaying ros messages in terminal, I think its a really neat tool and would be open to porting it to ROS2 for inclusion in the core `ros2cli` repo if the maintainers here find it valuable.

I'd take responsibility maintaining it, perhaps along with the original author @dheera 

https://github.com/dheera/rosshow

dirk-thomas, 2019-07-31:
Having `rosshow` available in ROS 2 would be great. It doesn't have to be part of the `ros2cli` repo for that. We usually suggest to keep the code in the existing repository and if necessary create a branch for the ROS 2 work in the same repo. That makes it easier for individual maintainers to keep track of tickets, maintain them on their own terms as well as cut releases independently.

The tool could either keep its current standalone executable name `rosshow` or integrate as a new command into `ros2 *`, e.g. `ros2 show` (or `ros2 render-msg`, just an idea to avoid ambiguity). The commands listed under `ros2 *` are Python entry points - so any Python package can contribute to them.

SteveMacenski, 2019-07-31:
Awesome, sounds good. I think then my action items:
- port to ROS2 (or support the improvement of a port)
- write a wrapper for this repo so we can call via `ros2 *` which I think is partially the value proposition of the work to have it out of the box

dirk-thomas, 2019-07-31:
> write a wrapper for this repo so we can call via `ros2 *`

You don't need any wrapper or anything in this repo in order to be able to call `ros2 show`. The package only needs to declare `show` in its setup file like this:

```
entry_points={
    'ros2cli.command': [
        'show = rosshow:ShowCommand',
    ],
}
```

and implement the interface of this extension point: https://github.com/ros2/ros2cli/blob/f8afbed18c7d299981b73c95ba26e7f386da9dc4/ros2cli/ros2cli/command/__init__.py#L24

SteveMacenski, 2019-07-31:
Ah I see, that's neat, good to know. 

:+1:

artivis, 2019-08-01:
+1 for getting `rosshow` out to the masses. However this can be achieved without actually landing the code in `ros2cli` repo, it could e.g. be added to the list of [ROS2 repos](https://github.com/ros2/ros2/blob/master/ros2.repos). Let's see what the maintainers prefer tho.
@SteveMacenski If necessary you may find a minimal example of creating a new ROS 2 CLI command/verb at [`ros2hellocli` repo](https://github.com/artivis/ros2hellocli). I'm also preparing a small tuto accompanying this example.

dirk-thomas, 2019-08-01:
Also the inclusion into the ROS 2 repos file isn't important. The most important thing is to release the package with `bloom` into a ROS 2 distro.

SteveMacenski, 2019-08-01:
@artivis this is a good point. @dirk-thomas sure, it's able for someone to get it, but unless its listed as a dependency to a core ros2 package, or in the ros2 repos file, I'd say its "available" to the masses but its not default to a point where I can just expect any developer from now on to know it because its one of the auto-complete options of the `ros2 *` CLI tools installed.

This isn't a project I wasn't a part of developing but I find it so incredibly useful I really think it should be shipped with a base install.

dirk-thomas, 2019-08-01:
> it should be shipped with a base install.

What do you mean by that? The archives of prebuild binaries we provide for each platform? The from source instructions? The Debian packages after installing what package?

sloretz, 2019-08-01:
> I really think it should be shipped with a base install.

Meaning, you think it should be [included in `ros-*-ros-core`](https://github.com/ros2/variants/blob/master/ros_core/package.xml#L51-L63)? That only requires a `bloom` release and a PR to [ros2/variants](https://github.com/ros2/variants), but it'd be nice if the repo were moved into an org like https://github.com/ros2 .

SteveMacenski, 2019-08-01:
I meant when you install ros core 
```
apt-cache show ros-dashing-ros-core

Depends: ros-dashing-ament-cmake, ros-dashing-ament-cmake-auto, ros-dashing-ament-cmake-gmock, ros-dashing-ament-cmake-gtest, ros-dashing-ament-cmake-pytest, ros-dashing-ament-cmake-ros, ros-dashing-ament-index-cpp, ros-dashing-ament-index-python, ros-dashing-ament-lint-auto, ros-dashing-ament-lint-common, ros-dashing-class-loader, ros-dashing-common-interfaces, ros-dashing-pluginlib, ros-dashing-rcl-lifecycle, ros-dashing-rclcpp, ros-dashing-rclcpp-lifecycle, ros-dashing-rclpy, ros-dashing-ros-environment, ros-dashing-ros2action, ros-dashing-ros2component, ros-dashing-ros2launch, ros-dashing-ros2lifecycle, ros-dashing-ros2msg, ros-dashing-ros2multicast, ros-dashing-ros2node, ros-dashing-ros2param, ros-dashing-ros2pkg, ros-dashing-ros2run, ros-dashing-ros2service, ros-dashing-ros2srv, ros-dashing-ros2topic, ros-dashing-rosidl-default-generators, ros-dashing-rosidl-default-runtime, ros-dashing-sros2, ros-dashing-sros2-cmake, ros-dashing-ros-workspace
```

Which I assumed was tied to this repo's metapackage `package.xml`, which is as sloretz points out, is incorrect. 

Yes @sloretz that's a good point, where its _actually_ defined :+1: . That's more or less what I meant about either including in the repos file or in the package.xml of a core package as a dependency for which that list defines (I had thought of dumping the dependency in here, but makes more sense there).

I understand now, thanks. I thought the packages for `core` installs were hand curated from `ros2` org repos, I didn't realize that's what the `variants` repo accomplishes. Thanks! I can take it from there. It seems the author is pretty open to having it relocated, most the reason I thought it should just go in here - but a separate repo would be fine as well. 

Quick question: Why not list all the `ros2cli` packages in the `ros2cli` metapackage XML as `exec_depend`s, then in the variant repo `package.xml` you can just list `ros2cli` and it will grab the rest?

dirk-thomas, 2019-08-01:
The packages included in the "core" variant are neither defined by the inclusion into specific repositories not by the inclusion into the `ros2.repos` file. They are defined in [REP 2001](http://www.ros.org/reps/rep-2001.html). So in order to include a package into "core" you should propose the addition there for future ROS distros. The lower level the variant is the higher the expectation on the maintainer will be to release the package promptly when new distributions are being bootstrapped.

Adding more and more packages into a single repo is a scalability problem: issues are mixed, maintainer roles are unclear, all packages share a version number and must be released together, etc.

The `ros2.repos` file is used by our monolithic CI on ci.ros2.org and currently to produce the binary archive for all platforms. But that is hopefully a short term thing since we really want to switch that to use `rosinstall_generator`. So I would say you should expect the `ros2.repos` file to go away at some point.

> Why not list all the `ros2cli` packages in the `ros2cli` metapackage XML as `exec_depends`, then in the variant repo `package.xml` you can just list `ros2cli` and it will grab the rest?

The `ros2cli` package is not a metapackage. It contains the framework part of the command line tools and the main executable. So it can't depend on the packages actually providing command since those are depending on `ros2cli`. In general we don't use metapackages much morein ROS 2 - even the special package type doesn't exist as a concept in ROS 2 (though you can achieve the same with an empty package which just has exec dependencies).

SteveMacenski, 2019-08-01:
Got it, thanks for the full explanation

SteveMacenski, 2019-08-01:
I'll close this ticket then and circle back in variants repo when I have something worth putting forward for eloquent and forward

closed 2019-08-01 by SteveMacenski