ros2/ros2cli #313: [test_ros2cli] Action CLI test failures on Windows

jacobperron, 2019-08-09:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - source
- Version or commit hash:
  - master

Follow-up from #310 

E.g. https://ci.ros2.org/job/ci_windows/7715/

ivanpauno, 2019-08-09:
I think something like https://github.com/ros2/ros2cli/blob/0ce8efa618c31a82f4f49f152a7347de9d234b04/ros2run/ros2run/api/__init__.py#L37-L43 is missing in https://github.com/ros2/launch_ros/blob/7e4f43156c729839cc8e6dd986f32bcac169a25b/launch_ros/launch_ros/substitutions/executable_in_package.py#L77-L82.

Probably, an utility providing that functionality could be added to `ament_index_python`.

ivanpauno, 2019-08-12:
> I think something like
>
> https://github.com/ros2/ros2cli/blob/0ce8efa618c31a82f4f49f152a7347de9d234b04/ros2run/ros2run/api/__init__.py#L37-L43
>
> is missing in [...]

I'm wrong. `Node` action was handling this correctly.

closed 2019-08-13 by ivanpauno