ros2/ros2cli #316: [ROS2][topic][pub] it does NOT support the size of array which is larger then 32

hyunseok-yang, 2019-08-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Binaries(Debian packages)
- Version or commit hash:
  - dashing
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
I tried to call topic pub through below command.
**ros2  topic pub -1 --qos-reliability reliable /initialpose geometry_msgs/PoseWithCovarianceStamped "{header: {frame_id: map}, pose: {pose: {position: {x: 0.0, y: 0.0, z: 0.0},              orientation: {x: 0.0, y: 0.0, z: 0, w: 1.0000000}},        covariance: [0.25, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.25, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.06853891945200942]}}"**

But it does not work properly with dashing version. 
(it worked with crystal version)

```

#### Expected behavior
like this
ros2 topic pub -1 --qos-reliability reliable /initialpose geometry_msgs/PoseWithCovarianceStamped "{header: {frame_id: 'map'}}"
**publisher: beginning loop
publishing #1: geometry_msgs.msg.PoseWithCovarianceStamped(header=std_msgs.msg.Header(stamp=builtin_interfaces.msg.Time(sec=0, nanosec=0), frame_id='map'), pose=geometry_msgs.msg.PoseWithCovariance(pose=geometry_msgs.msg.Pose(position=geometry_msgs.msg.Point(x=0.0, y=0.0, z=0.0), orientation=geometry_msgs.msg.Quaternion(x=0.0, y=0.0, z=0.0, w=0.0)), covariance=array([0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.,
       0., 0.])))**

#### Actual behavior
But,
I got this

**Failed to populate field: maximum supported dimension for an ndarray is 32, found 36**

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

sloretz, 2019-08-19:
It looks like [rosidl_runtime_py.set_message:set_message_fields](https://github.com/ros2/rosidl_python/blob/91748e2231a854f5fab95014f0e2954fdeb5b42f/rosidl_runtime_py/rosidl_runtime_py/set_message.py#L39-L43) needs a special case for `numpy.ndarray()`

```python3
from geometry_msgs.msg import PoseWithCovariance
from rosidl_runtime_py.set_message import set_message_fields
m = PoseWithCovariance()
set_message_fields(m, {'covariance': [0.] * 36})
```

Results in 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/ros/dashing/lib/python3.6/site-packages/rosidl_runtime_py/set_message.py", line 43, in set_message_fields
    value = field_type(field_value)
ValueError: sequence too large; cannot be greater than 32
```

The constructor of `ndarray()` needs the data passed in via the `buffer` keyword arg.

```python3
>>> len(numpy.ndarray(buffer=numpy.array([0.] * 4000), shape=(36,)))
36
>>> len(numpy.ndarray(buffer=numpy.array([0.] * 4000), shape=(4000,)))
4000
```

The first keyword arg is `shape`, which has a limit of 32 dimensions.

```python3
>>> numpy.ndarray([0.]*36)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: sequence too large; cannot be greater than 32
>>> numpy.ndarray([0.]*36, shape=(36,))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: Argument given by name ('shape') and position (1)
```

hyunseok-yang, 2019-08-20:
Yes, that's what I pointed out exactly :)
Is there any trick that I can utilize?

Or, let me know if you have a permanent solution for these situation :+1:

dirk-thomas, 2019-08-23:
Please see ros2/rosidl_python#81 for a proposed fix. It would be great if you could try it and comment if it resolves the problem for you.

hyunseok-yang, 2019-08-26:
> Please see [ros2/rosidl_python#81](https://github.com/ros2/rosidl_python/pull/81) for a proposed fix. It would be great if you could try it and comment if it resolves the problem for you.

@dirk-thomas 
Thank you!!  it worked as desired! :100: