ros2/ros2cli #321: [ros2component] find_container_node_names sometimes fails in the presence of other nodes

cottsay, 2019-08-24:
The `ros2component` API tests call `find_container_node_names`, which sometimes fails in the presence of other running tests:

```
________________________ test_find_container_node_names ________________________
test/test_api.py:30: in test_find_container_node_names
    assert len(find_container_node_names(
ros2component/api/__init__.py:204: in find_container_node_names
    services = get_service_info(node=node, remote_node_name=n.full_name)
../../../../install/ros2node/lib/python3.6/site-packages/ros2node/api/__init__.py:76: in get_service_info
    return get_topics(remote_node_name, node.get_service_names_and_types_by_node)
../../../../install/ros2node/lib/python3.6/site-packages/ros2node/api/__init__.py:59: in get_topics
    names_and_types = func(node.name, node.namespace)
../../../../install/rclpy/lib/python3.6/site-packages/rclpy/node.py:1505: in get_service_names_and_types_by_node
    capsule, node_name, node_namespace)
E   RuntimeError: Failed to get_service_names_and_types: Unable to find GUID for node , at /home/jenkins-agent/workspace/nightly_linux_release/ws/src/ros2/rmw_fastrtps/rmw_fastrtps_shared_cpp/src/rmw_node_info_and_types.cpp:87
```

On the CI farm, it seems to be common that this test runs at the same time as the `pendulum_control` tests, and it causes this test to fail. I was able to reproduce the behavior locally by running both tests in simultaneously in a loop.

ivanpauno, 2019-08-26:
I think, the problem is that we're first [getting a list of node names](https://github.com/ros2/ros2cli/blob/10c89eb1b6850d1e7fe21e6b5487b68114cf0059/ros2component/test/test_api.py#L24-L36), and then using it for [list services of those nodes](https://github.com/ros2/ros2cli/blob/10c89eb1b6850d1e7fe21e6b5487b68114cf0059/ros2component/ros2component/api/__init__.py#L192-L218). If the node has disappeared in the middle, `get_service_info` will fail.

The problem is not only the test, we're using the same logic in other places: https://github.com/ros2/ros2cli/blob/10c89eb1b6850d1e7fe21e6b5487b68114cf0059/ros2component/ros2component/verb/list.py#L41-L46.

IMO, `find_container_node_names` should be rewritten to handle the error nicely.

hidmic, 2019-08-26:
That is correct. I wonder if this is the only place where we have such races.

cottsay, 2019-09-16:
Is the resolution for this issue progressing? It's been the only nightly failure for Linux release/debug for several days now. It would be great to get those builds clean again.

If the fix is still a ways out, I'd like to reconsider isolating the domain ID of the test and adding another test case for this issue specifically, since the test is failing due to an issue it isn't actually looking for.

dirk-thomas, 2019-09-16:
> I'd like to reconsider isolating the domain ID of the test

As discussed on previous tickets there is no good approach to select a unique but no colliding domain id in our test infrastructure atm.

ivanpauno, 2019-09-16:
I'm working on this. I'll finish with it between today and tomorrow.

cottsay, 2019-09-16:
Thanks for the update!

PR #322 (2019-09-17): Handle find_container_node_names error
Fix #321.

There is a raise condition, as I described [here](https://github.com/ros2/ros2cli/issues/321#issuecomment-524834399).

The error I'm catching comes from here:
https://github.com/ros2/rclpy/blob/8da91cee54b68f1108ca017da885496ffb5ae16c/rclpy/src/rclpy/_rclpy.c#L3137-L3146

The problem is that I can be also hiding some other errors by doing this.
`rcl_get_service_names_and_types_by_node` is a wrapper of `rmw_get_service_names_and_types_by_node`.
`rmw_get_service_names_and_types_by_node` can fail not only because the node name wan't found, but als [... cut]
closed 2019-09-17 by ivanpauno