ros2/ros2cli #327: Ros2 doctor does not show middleware details

rotu, 2019-09-12:
## Feature request

#### Feature description

ROS2 doctor should list middleware name and library versions. This is useful troubleshooting info, and it would be great to have it available to paste into issue reports.

#### Implementation considerations
The names of the relevant libraries and system packages are not currently part of the RMW interface, and it may make sense to add some version info alongside `rmw_get_implementation_identifier`. Also, it’s important to communicate things like libopensplice67 vs libopensplice69, which I don’t think is exposed anywhere

rotu, 2019-09-12:
Based on a good idea by @SteveMacenski

claireyywang, 2019-09-19:
I'm adding RMW name for now. More detailed info will be added either after I get more familiar with the cpp code base or by community effort.