ros2/ros2cli #336: [ros2component] Parameters and remap rules cause log warning

jacobperron, 2019-09-19:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 5dc4d9bfe14a2a62667977bfaa52b2288af6c2f3
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

Load a standalone component and pass it a parameter or remap rule:
```
ros2 component standalone demo_nodes_cpp demo_nodes_cpp::Talker -p foo:=bar
```

```
ros2 component standalone demo_nodes_cpp demo_nodes_cpp::Talker -r chatter:=bar
```

#### Expected behavior

Parameter and remap options for `ros2 component standalone` work as documented without side-effects.

#### Actual behavior

The flags appear to work (ie. successfully set a parameter or remap a topic), but anything matching the pattern "X:=Y" is picked up additionally as a remap and we see a warning, e.g.:

```
[WARN] [rcl]: Found remap rule 'foo:=bar'. This syntax is deprecated. Use '--ros-args --remap foo:=bar' instead.
```

If we try to set a parameter with `-p foo:=bar` it will also remap any topic named `foo` to `bar` as well.

sloretz, 2019-09-23:
Does it still get interpreted as a remap rule when `--ros-args` is given?

```
ros2 component standalone demo_nodes_cpp demo_nodes_cpp::Talker --ros-args -p foo:=bar
```

jacobperron, 2019-09-23:
@sloretz Not sure. `ros2cli` won't let us get that far:

```
ros2: error: unrecognized arguments: --ros-args
```

I'm not sure if we want to allow `--ros-args` to be passed alonside the CLI tools, but if so we should probably ticket it.

clydemcqueen, 2019-12-19:
Also a bug for extra arguments, e.g.,
`ros2 component load /ComponentManager opencv_cam opencv_cam::ImageSubscriberNode -e use_intra_process_comms:=true`

commit 26619feb (2020-01-13): Avoid passing None to rclpy.init

Otherwise, all CLI arguments will be parsed as ROS arguments, which can lead
to rcl warnings or incorrect legacy remapping behavior.

This change does not pass arguments to rclpy.init from any of the CLI
tools, but it leaves the opportunity to do so in the future by setting the
parser argument 'argv'. For example, we could take the remaining arguments
and pass them to rclpy.init, similar to what is done in ros2run:

https://github.com/ros2/ros2cli/blob/4c5d9327026ecb2ea10a16b3429908b4f6f64ca6/ros2run/ros2run/command/run.py#L51-L53

Fixes #336.
PR #433 (2020-01-28): Avoid passing None to rclpy.init
Otherwise, all CLI arguments will be parsed as ROS arguments, which can lead
to rcl warnings or incorrect legacy remapping behavior.

This change does not pass arguments to rclpy.init from any of the CLI
tools, but it leaves the opportunity to do so in the future by setting the
parser argument 'argv'. For example, we could take the remaining arguments
and pass them to rclpy.init, similar to what is done in ros2run:

https://github.com/ros2/ros2cli/blob/4c5d9327026ecb2ea10a16b3429908b4f6f64ca6/ros2run/ros2run/command/run.py#L51-L53

Fixes #336.

* Linux [image]  [... cut]
commit 2b6286cd (2020-01-28): Avoid passing None to rclpy.init (#433)

Otherwise, all CLI arguments will be parsed as ROS arguments, which can lead
to rcl warnings or incorrect legacy remapping behavior.

This change does not pass arguments to rclpy.init from any of the CLI
tools, but it leaves the opportunity to do so in the future by setting the
parser argument 'argv'. For example, we could take the remaining arguments
and pass them to rclpy.init, similar to what is done in ros2run:

https://github.com/ros2/ros2cli/blob/4c5d9327026ecb2ea10a16b3429908b4f6f64ca6/ros2run/ros2run/command/run.py#L51-L53

Fixes #336.
commit d812b2f8 (2020-02-09): Avoid passing None to rclpy.init (#433) (#449)

Otherwise, all CLI arguments will be parsed as ROS arguments, which can lead
to rcl warnings or incorrect legacy remapping behavior.

This change does not pass arguments to rclpy.init from any of the CLI
tools, but it leaves the opportunity to do so in the future by setting the
parser argument 'argv'. For example, we could take the remaining arguments
and pass them to rclpy.init, similar to what is done in ros2run:

https://github.com/ros2/ros2cli/blob/4c5d9327026ecb2ea10a16b3429908b4f6f64ca6/ros2run/ros2run/command/run.py#L51-L53

Fixes #336.
closed 2020-01-28 by jacobperron