ros2/ros2cli #343: ros2doctor: add support for warning on out of date packages

tfoote, 2019-09-30:
A common error that people have is that they may have old our outdated packages on their system. This is most common with build pipeline tools, but can also have an effect due to a partial install or update of their debian packages. 

ros2doctor could collect the list of all local packages in the workspace with their version number and then check against a declared rosdistro for the released version number. And if the latest version number and the version in the rosdistro do not match it should add it to the warnings. 

We don't have the rosdistro cache for our latest bootstrap packages, but there may be an approach that would be reasonable to query pypi for the latest version or the apt servers, or possibly a github API. Bloom has logic like this to warn if it's not the latest version that could potentially be replicated.

In the report mode it would be valuable to be able to export the list of packages present on the system as well as their version number. 

In report mode another extension that would be valuable would be the `vcs export --exact` of the workspace for easier reproduction/debugging. As well as potentially listing the exact debian packages (with build numbers) that have been installed for easier reproduction.

claireyywang, 2019-10-08:
I'm using `vcs export --exact` API to get local package information, but it will make this check directory sensitive, since `vcs` only lists packages installed within the directory it is called. `roswtf` also had the same property, so I guess it is ok for `ros2doctor` to be the same?

dirk-thomas, 2019-10-08:
How do you know the location of the sources of the packages? The same problem was discussed in ros2/ros2cli#75.

claireyywang, 2019-10-08:
Good question. I think that I'm facing the same dilemma as in the discussion. `vcstool` only works if the workspace it's called in contains source, thus this check would not work in some cases. I am unable to figure out a good solution at the moment, but open to suggestions.

dirk-thomas, 2019-10-08:
In a similar case we are using `colcon` to actually find the source directory of packages (see colcon/colcon-cd#1). Not sure if it is desired to depend on that here though...

claireyywang, 2019-12-26:
closing since pr is merged

closed 2019-12-26 by claireyywang