ros2/ros2cli #344: Give actions their own category in ros2 node info results

maryaB-osr, 2019-10-02:
## Feature request

#### Actions in their own category with `node info`

`ros2 node info` returns a list of subscribers, publishers, and services, with actions in hidden namespaces under the publishers and services categories.

```
$ ros2 node info /fibonacci_action_server
/fibonacci_action_server
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /fibonacci/_action/feedback: action_tutorials_interfaces/action/Fibonacci_FeedbackMessage
    /fibonacci/_action/status: action_msgs/msg/GoalStatusArray
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Services:
    /fibonacci/_action/cancel_goal: action_msgs/srv/CancelGoal
    /fibonacci/_action/get_result: action_tutorials_interfaces/action/Fibonacci_GetResult
    /fibonacci/_action/send_goal: action_tutorials_interfaces/action/Fibonacci_SendGoal
    /fibonacci_action_server/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /fibonacci_action_server/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /fibonacci_action_server/get_parameters: rcl_interfaces/srv/GetParameters
    /fibonacci_action_server/list_parameters: rcl_interfaces/srv/ListParameters
    /fibonacci_action_server/set_parameters: rcl_interfaces/srv/SetParameters
```

It would be nice if actions had their own category in these results. It would benefit the clarity of the tutorials/learning process if actions could be visually distinct like the other node elements, for consistency.

```
$ ros2 node info /fibonacci_action_server
/fibonacci_action_server
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Services:
    /fibonacci_action_server/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /fibonacci_action_server/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /fibonacci_action_server/get_parameters: rcl_interfaces/srv/GetParameters
    /fibonacci_action_server/list_parameters: rcl_interfaces/srv/ListParameters
    /fibonacci_action_server/set_parameters: rcl_interfaces/srv/SetParameters
  Actions:
    /fibonacci/feedback: action_tutorials_interfaces/action/Fibonacci_FeedbackMessage
    /fibonacci/status: action_msgs/msg/GoalStatusArray    
    /fibonacci/cancel_goal: action_msgs/srv/CancelGoal
    /fibonacci/get_result: action_tutorials_interfaces/action/Fibonacci_GetResult
    /fibonacci/send_goal: action_tutorials_interfaces/action/Fibonacci_SendGoal

```

maryaB-osr, 2019-10-02:
@jacobperron

jacobperron, 2019-10-02:
> EDIT: I misunderstood what's going on here, so my "solution" isn't exactly right, but the feature request is the same - actions should have their own category

Right, the output should not include the hidden topics/services at all and should look like:
```
$ ros2 node info /fibonacci_action_server
/fibonacci_action_server
  Subscribers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Services:
    /fibonacci_action_server/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /fibonacci_action_server/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /fibonacci_action_server/get_parameters: rcl_interfaces/srv/GetParameters
    /fibonacci_action_server/list_parameters: rcl_interfaces/srv/ListParameters
    /fibonacci_action_server/set_parameters: rcl_interfaces/srv/SetParameters
  Actions:
    /fibonacci: action_tutorials_interfaces/action/Fibonacci
```

PR #345 (2019-10-03): ros2node info: add action category

Resolves #344 

Adds separate action category to `ros2 node info`. Output looks like 
```
clairewang@karr:~/ros2_ws/src/ros2/ros2cli/ros2node$ ros2 node info /fibonacci_action_server /fibonacci_action_server
  Subscribers:

  Publishers:
    /fibonacci/_action/feedback: action_tutorials_interfaces/action/Fibonacci_FeedbackMessage
    /fibonacci/_action/status: action_msgs/msg/GoalStatusArray
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Services:
    /fibonacci/_action/cancel_goal: action_msgs/srv/CancelGoal
  [... cut]
closed 2019-10-03 by claireyywang