ros2/ros2cli #347: [ros2topic] ros2 topic list does not have a -v/--verbose option

wjwwood, 2019-10-03:
## Feature request

#### Feature description

In ROS 1 there was a `-v` option which listed out data like this (from https://wiki.ros.org/ROS/Tutorials/UnderstandingTopics#Using_rostopic_list):

```
Published topics:
 * /turtle1/color_sensor [turtlesim/Color] 1 publisher
 * /turtle1/cmd_vel [geometry_msgs/Twist] 1 publisher
 * /rosout [rosgraph_msgs/Log] 2 publishers
 * /rosout_agg [rosgraph_msgs/Log] 1 publisher
 * /turtle1/pose [turtlesim/Pose] 1 publisher

Subscribed topics:
 * /turtle1/cmd_vel [geometry_msgs/Twist] 1 subscriber
 * /rosout [rosgraph_msgs/Log] 1 subscriber
```

This doesn't exist in ROS 2 yet, and it would be useful (for that tutorial and in general) for `ros2 topic list` as well.

evshary, 2019-10-04:
I study the issue recently. I'd like to work on it.

matthews-jca, 2021-07-13:
Going through some "good first issues" and noted that this issue appears to have been fixed and merged.

clalancette, 2021-07-13:
Yep, very good point.  I'm going to close this out.

closed 2021-07-13 by clalancette