ros2/ros2cli #348: [ros2msg] does not show "expanded" message definition

wjwwood, 2019-10-04:
## Feature request

#### Feature description

In ROS 1 `rosmsg show` would show you the recursively expanded message fields, whereas in ROS 2 at the moment `ros2 msg show` only shows you the fields of the immediate message you specify.

For example, in ROS 1 (from https://wiki.ros.org/ROS/Tutorials/UnderstandingTopics#Using_rostopic_type):

```
$ rosmsg show geometry_msgs/Twist
geometry_msgs/Vector3 linear
  float64 x
  float64 y
  float64 z
geometry_msgs/Vector3 angular
  float64 x
  float64 y
  float64 z
```

And the equivalent in ROS 2:

```
% ros2 msg show geometry_msgs/msg/Twist
# This expresses velocity in free space broken into its linear and angular parts.

Vector3  linear
Vector3  angular
```

It does, however, show the comments in the message file, which may be nice to have some times.

Also, as a side note, this command takes almost 2 seconds to run on my macbook, but I'm not sure why...

sonpeppers, 2019-12-30:
This seems like a good enhancement for beginner, I'd like to take this one

jacobperron, 2020-01-17:
@sonpeppers Great! Let me know if you have any questions.

audrow, 2020-05-21:
@sonpeppers How is this going?  I'm thinking about giving it a try.

PR #524 (2020-06-05): [ros2interface] show "expanded" message definition
Fixes #348
closed 2020-06-05 by audrow