ros2/ros2cli #349: [ros2msg] does not allow input from stdin for `show`

wjwwood, 2019-10-04:
## Feature request

#### Feature description

In this ROS 1 tutorial (https://wiki.ros.org/ROS/Tutorials/UnderstandingTopics#Using_rostopic_hz) there is a shell one-liner:

```
$ rostopic type /turtle1/cmd_vel | rosmsg show
```

Which then shows the structure of the topic type for the `/turtle1/cmd_vel`.

It would be nice to have this ability in the ROS 2 `ros2 msg show` command line tool too.

sharminramli, 2019-10-21:
I'd like to work on this, but it looks like ros2msg was deprecated in #363. Can I go ahead and look into implementing this feature in ros2interface if it's still relevant?

jacobperron, 2019-10-21:
@sharminramli Thanks for volunteering to look into this!
Correct, ros2msg is deprecated, but I believe the feature is still relevant to ros2interface. I'll change the title and description of this ticket accordingly.

sharminramli, 2020-11-06:
this can probably be closed?

clalancette, 2020-11-06:
Yeah, good call.  Closing this out.

closed 2020-11-06 by clalancette