ros2/ros2cli #350: [ros2srv] does not allow input from stdin for `show`

maryaB-osr, 2019-10-05:
#### Feature description

In this ROS 1 tutorial (http://wiki.ros.org/ROS/Tutorials/UnderstandingServicesParams#rosservice_call) there is a shell one-liner:
```
$ rosservice type /spawn | rossrv show
```
Which then shows the structure of the service type for the `/spawn` service

It would be nice to have this ability in the ROS 2 `ros2 srv show` command line tool too.

rummanwaqar, 2019-10-05:
Hey @maryaB-osr I am doing hacktoberfest and would like to work on this issue.

maryaB-osr, 2019-10-05:
@rummanwaqar Great, thanks!

jacobperron, 2019-10-21:
@rummanwaqar FYI, we've recently deprecated the package `ros2srv`, instead preferring `ros2interface` (#363). If you are still working on this (or plan to), please refer to the equivalent ticket for `ros2interface`: https://github.com/ros2/ros2cli/issues/349
Note, there might be someone else working on this so please try to coordinate on that ticket to avoid duplicate work.

closed 2019-10-21 by jacobperron