ros2/ros2cli #369: ros2pkg create package_name in wrong order

mabelzhang, 2019-10-17:
## Bug report

Ordering of `package_name` in help message of `ros2 pkg create` is wrong. It should precede all optional args, instead of at the end (which doesn't work).

I thought I could try to fix this quickly but got as far as `ros2pkg/ros2pkg/verb/create.py`, and looks like it's just `argparse`. So I'm not sure where this issue originates.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - From source
- Version or commit hash:
  - Latest in dashing branch ca3176188a0e0a75a9dee8915ff8f3adb1eab13b
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
$ ros2 pkg create --build-type ament_cmake --dependencies rclcpp std_msgs tmp
```

#### Expected behavior
Package `tmp` created with dependencies `rclcpp` and `std_msgs`.

#### Actual behavior
Error message:
```
usage: ros2 pkg create [-h] [--package_format {2,3}]
                       [--description DESCRIPTION] [--license LICENSE]
                       [--destination-directory DESTINATION_DIRECTORY]
                       [--build-type {cmake,ament_cmake}]
                       [--dependencies DEPENDENCIES [DEPENDENCIES ...]]
                       [--maintainer-email MAINTAINER_EMAIL]
                       [--maintainer-name MAINTAINER_NAME]
                       [--cpp-node-name CPP_NODE_NAME]
                       [--cpp-library-name CPP_LIBRARY_NAME]
                       package_name
ros2 pkg create: error: the following arguments are required: package_name
```

Note `package_name` is at the end of the list, but that format doesn't work in practice.

#### Additional information

The package_name arg has to be in front:
```
$ ros2 pkg create tmp --build-type ament_cmake --dependencies rclcpp std_msgs    
```

Help message should have `package_name` preceding all the optional args to reflect this.

Karsten1987, 2019-10-17:
I thought this is just how `argparse` works in a way that it's greedy with the optional args.
I would assume that `ros2 pkg create tmp --build-type ament_cmake --dependencies rclcpp std_msgs  -- tmp` would work.

mabelzhang, 2019-10-26:
Confirming that works for me. Never knew about using the empty `--` after a list of args...
Closing.

closed 2019-10-26 by mabelzhang