ros2/ros2cli #378: ros2doctor: failed network check due to missing dependency

claireyywang, 2019-10-23:
## Bug report

**Required Info:**

- Operating System:
  - Linux Ubuntu 18.04 
- Installation type:
  - Source 
- Version or commit hash:
  - Master 
- DDS implementation:
  - All 
- Client library (if applicable):
  - N/A 

#### Steps to reproduce issue

```
ros2 doctor 
```

#### Expected behavior
Run a list of checks and output check results.
```
$ ros2 doctor 
/home/osrf/ros2_ws/build/ros2doctor/ros2doctor/api/format.py: 76: UserWarning: Distribution eloquent is not fully supported or tested. To get more consistent features, download a stable version at https://index.ros.org/doc/ros2/Installation/

All 3 checks passed
```
#### Actual behavior
Unix machines without `net-tools` installed or Windows machines have failed network check.
```
/home/claire/ros2_ws/build/ros2doctor/ros2doctor/api/format.py: 76: UserWarning: ERROR: No loopback IP address is found.
/home/claire/ros2_ws/build/ros2doctor/ros2doctor/api/format.py: 76: UserWarning: Only loopback IP address is found.
/home/claire/ros2_ws/build/ros2doctor/ros2doctor/api/format.py: 76: UserWarning: No multicast IP address is found.
/home/claire/ros2_ws/build/ros2doctor/ros2doctor/api/format.py: 76: UserWarning: Distribution eloquent is not fully supported or tested. To get more consistent features, download a stable version at https://index.ros.org/doc/ros2/Installation/

1/3 checks failed

Failed modules are  network

```
Need to add package dependency on `net-tools`, specifically the `ifconfig` terminal tool. Need an alternative tool for windows system.

mjcarroll, 2019-10-24:
It appears that ifcfg should be cross-platform and work on Windows and macOS: https://github.com/ftao/python-ifcfg

claireyywang, 2019-10-24:
I found that the python package itself is unable to get the flag content `flags=4163<UP,BROADCAST,RUNNING,MULTICAST>` which is essentially the network check, and instead output the check as failed even though some machines have loopback and multicast set up. It seems that `ifcfg` relies on the terminal tool `ifconfig` to some degree. To resolve this issue, I need to either find an alternative or add `ifconfig` as a dependency.

claireyywang, 2019-11-11:
dependency added in rosdep https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L4252

closed 2019-11-11 by claireyywang