ros2/ros2cli #382: [ros2interface]: "interface list" verb doesn't display intermediate directory

clalancette, 2019-10-23:
If I run `ros2 msg list`, I get a (heavily edited) output that looks like:

```
turtlesim/msg/Color
turtlesim/msg/Pose
```

If I run `ros2 interface list`, I get a (heavily edited) output that looks like:

```
Messages:
    turtlesim/Color
    turtlesim/Pose
```

While the "Messages" category is explicit that these are messages, I think that it would be best to be consistent here and always print the "intermediate" `msg` here.

jacobperron, 2019-10-24:
This actually due to a bug in the rosidl_runtime_py functions being called by the list verb.

jacobperron, 2019-10-24:
See https://github.com/ros2/rosidl_runtime_py/pull/6 for a fix.

jacobperron, 2019-10-24:
Consequently, this also fixes a similar issue with completion for the `show` verb.

closed 2019-10-24 by jacobperron

hidmic, 2019-10-24:
As I said on https://github.com/ros2/rosidl_runtime_py/pull/6, I think `ros2interface` tests have not been properly updated :)

jacobperron, 2019-10-24:
> As I said on ros2/rosidl_runtime_py#6, I think ros2interface tests have not been properly updated :)

Good catch, my bad. I should have included ros2interface in the CI build...

See https://github.com/ros2/ros2cli/pull/386 for a fix.