ros2/ros2cli #411: [ros2topic] info verb should support hidden topics

jacobperron, 2019-12-02:
## Feature request

#### Feature description

We can list hidden topics with the flag `--include-hidden-topics`

    ros2 topic list --include-hidden-topics

But we can't get info about any hidden topics.

For example, start publishing to a hidden topic:

    ros2 topic pub /_chatter std_msgs/msg/String "data: hello" 

And in another terminal, attempt to get info for the hidden topic:

    ros2 topic info /_chatter

We get the error:

    Unknown topic '/_chatter'

But I would expect the output to be:

    Type: std_msgs/msg/String
    Publisher count: 1
    Subscriber count: 0

evshary, 2019-12-15:
This feature can be supported by adding argument `include_hidden_topics=True`.
However I'm not sure whether it's fine to show hidden topic directly or it's necessary to add extra option, like `--include-hidden-topics`.

jacobperron, 2019-12-16:
Closed by #423

closed 2019-12-16 by jacobperron