ros2/ros2cli #417: node info CLI bug: 'Namespace' object has no attribute 'include_hidden'

IanTheEngineer, 2019-12-04:
Bug report
When running `ros2 node info /my_node`, I see a traceback at the end of the Publishers section due to 'include_hidden' attribute being missing. I would hazard a guess that this is related to https://github.com/ros2/ros2cli/pull/409

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - ROS Dashing-Testing Binaries: ros2node-0.7.8
- Version or commit hash:
  - 0.7.8 of ros2node
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

```
$ ros2 node info /my_node
/my_node
  Subscribers:
    /topic1: std_msgs/msg/String
    /topic2: std_msgs/msg/Float32
    /parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /topic_a: std_msgs/msg/String
    /topic_b: std_msgs/msg/Float32
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
Traceback (most recent call last):
  File "/opt/ros/dashing/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.7.8', 'console_scripts', 'ros2')()
  File "/opt/ros/dashing/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/dashing/lib/python3.6/site-packages/ros2node/command/node.py", line 39, in main
    return extension.main(args=args)
  File "/opt/ros/dashing/lib/python3.6/site-packages/ros2node/verb/info.py", line 54, in main
    node=node, remote_node_name=args.node_name, include_hidden=args.include_hidden)
AttributeError: 'Namespace' object has no attribute 'include_hidden'
```

#### Expected behavior
No traceback at the end of the command output.
#### Actual behavior
`Traceback... AttributeError: 'Namespace' object has no attribute 'include_hidden'`

dirk-thomas, 2019-12-04:
Thanks for the report. #418 should address this regression.

hidmic, 2020-04-27:
Can this be closed now?

PR #418 (2019-12-05): partially revert [dashing backport] add service clients to ros2node info
Fixes #417 by reverting #409 and only adding the API for cross-distro compatibility.

The other parts (fetching client information as well as the hidden topic option require other changes which are not available in Dashing).
closed 2020-04-27 by dirk-thomas