ros2/ros2cli #419: Improve ros2 daemon API

hidmic, 2019-12-05:
## Feature request

#### Feature description

To decouple CLI commands from node discovery latencies, we currently spawn a daemon process that caches the ROS graph, speeding up queries. But we do so sparsely. Several verbs implementations -- [`ros2 security generate policy`](https://github.com/ros2/sros2/blob/083a850efa5cb8900b3fa8d8a9ec9460e6d7e034/sros2/sros2/verb/generate_policy.py#L119-L130), [`ros2 node info`](https://github.com/ros2/ros2cli/blob/19a491851efb043fac6b7e740c2838d6e1fb15e6/ros2node/ros2node/verb/info.py#L47-L52) and [`ros2 param list`](https://github.com/ros2/ros2cli/blob/19a491851efb043fac6b7e740c2838d6e1fb15e6/ros2param/ros2param/verb/list.py#L46-L59), just to name a few -- first use the daemon via `NodeStrategy` and then use `DirectNode`, effectively racing against discovery (e.g. the node in the daemon may have discovered a remote node already, but the node created by the CLI may have not _yet_). 

Thus, we should use the daemon correctly, potentially adding support for a larger subset of the ROS graph API in the process, and/or spin long enough so that discovery can take place.

jacobperron, 2020-08-05:
@hidmic The related PRs have been merged. Is there still work to do to resolve this ticket?

hidmic, 2020-10-21:
I believe not. Thanks for bumping @jacobperron.

closed 2020-10-21 by hidmic