ros2/ros2cli #424: ros2 commands are very slow

christianrauch, 2019-12-18:
## Bug report
The `ros2` command is very slow (more than 100 times slower than comparable ROS1 commands). E.g. on a Raspberry Pi 2, showing the help via `ros2 run -h` takes more than 7 seconds.

**Required Info:**
- Operating System: Ubuntu 18.04
- Installation type: binaries
- Version or commit hash: eloquent
- Client library (if applicable): rclpy

#### Steps to reproduce issue
1. install ROS2 on Ubuntu on a Raspberry Pi 2
2. source and run `time ros2 run -h`
3. example timing:
```
real    0m7,423s
user    0m6,863s
sys     0m0,559s
```

#### Expected behavior
The help text show be shown immediately (less than 0.1 second).

#### Actual behavior
Showing the help text will take more than 7 seconds!

#### Additional information
On a more powerful PC, the `ros2` command will take e.g. `0m0,808s`. The comparable command on ROS1 (`time rosrun -h`) will take `0m0,006s` (i.e. less than 10ms). The ROS2 version of these commands is more than 100 times slower than the ROS1 commands. Showing help or other generic information should not take more than 100ms.

wjwwood, 2020-01-02:
`rosrun` vs `ros2 run` is not a very fair comparison, because in that particular case we had to convert from bash to python. And I don't think there's much to be done about that technical point. However, it still may be slower than it needs to be, or was in ROS 1.

A better comparison would be between `rostopic -h` and `ros2 topic -h`. Maybe you care to redo your analysis on those tools?

wjwwood, 2020-01-02:
I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2020-01-02 by wjwwood

dirk-thomas, 2020-01-18:
@christianrauch Please see #436 which improves the performance a bit (something between 20-50% depending on the command). While that doesn't get us to the time of native shell functions it is at least a step in the right direction.