ros2/ros2cli #425: [ros2doctor] Python exception thrown when following tutorial.

LoyVanBeek, 2019-12-23:
## Bug report
Running `ros2 doctor --report` in a Docker container fails with a Python stack trace

**Required Info:**

- Operating System:
  - Ubuntu 18.04.3 in `ros:eloquent` docker image
- Installation type:
  - Docker image, not sure how the docker image itself is built
- Version or commit hash:
  - 0.8.6 (as found by `dpkg -S ros2cli`, there is no `-v` flag yet)
- DDS implementation:
  - Fast-RTPS (nothing installed additional to what's in the docker image)

#### Steps to reproduce issue
I was following https://index.ros.org/doc/ros2/Tutorials/Getting-Started-With-Ros2doctor/#id6 in said docker image:
```
ros2 doctor
ros2 doctor --report
```

#### Expected behavior
An error information about network configuration

#### Actual behavior
```
root@5832e3497fea:~/ros2/dev_ws# ros2 doctor
/opt/ros/eloquent/lib/python3.6/site-packages/ros2doctor/api/format.py: 76: UserWarning: ERROR: No loopback IP address is found.
/opt/ros/eloquent/lib/python3.6/site-packages/ros2doctor/api/format.py: 76: UserWarning: Only loopback IP address is found.
/opt/ros/eloquent/lib/python3.6/site-packages/ros2doctor/api/format.py: 76: UserWarning: No multicast IP address is found.
/opt/ros/eloquent/lib/python3.6/site-packages/ros2doctor/api/format.py: 76: UserWarning: Publisher without subscriber detected on /topic.

1/3 checks failed

Failed modules: network
root@5832e3497fea:~/ros2/dev_ws# ros2 doctor --report

  NETWORK CONFIGURATION
Traceback (most recent call last):
  File "/opt/ros/eloquent/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.8.6', 'console_scripts', 'ros2')()
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2cli/cli.py", line 69, in main
    rc = extension.main(parser=parser, args=args)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2doctor/command/doctor.py", line 45, in main
    format_print(report_obj)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2doctor/api/format.py", line 27, in format_print
    padding_num = compute_padding(report.items)
  File "/opt/ros/eloquent/lib/python3.6/site-packages/ros2doctor/api/format.py", line 40, in compute_padding
    check_items = list(zip(*report_items))[0]  # get first elements of tuples
IndexError: list index out of range
root@5832e3497fea:~/ros2/dev_ws#
```

### Additional information
Output of `ifconfig`, inside the docker container
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 172.17.0.2  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:ac:11:00:02  txqueuelen 0  (Ethernet)
        RX packets 14832  bytes 26739102 (26.7 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 17365  bytes 2222044 (2.2 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 6892  bytes 1703199 (1.7 MB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 6892  bytes 1703199 (1.7 MB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

claireyywang, 2019-12-26:
I reproduced this error by downloading a new ros:eloquent image and there is no `ifconfig` command found. I was able to fix it by running `apt-get update & apt-get install net-tools`. Can you try that and see if that works?

LoyVanBeek, 2019-12-28:
Confirmed, manually installing ifconfig does fix the error indeed.

claireyywang, 2019-12-29:
closing since issue is fixed.

closed 2019-12-29 by claireyywang

dirk-thomas, 2020-01-02:
Manually installing `net-tools` doesn't seem like a valid fix for this problem. When the user installs the Debian package it should also install necessary dependencies and the user shouldn't see an exception when invoking `ros2 doctor` without manually installing other packages.

LoyVanBeek, 2020-01-03:
The problem is with `python3-ifcfg` rather than `ros2 doctor` maybe, so I proposed a [PR](https://github.com/ftao/python-ifcfg/pull/45) there. No change to ros2 doctor is required I think because it already handles errors like this

With https://github.com/ftao/python-ifcfg/pull/45, the output in the docker container is

```
root@83317aab73fc:/python-ifcfg/src/ifcfg# ros2 doctor --report
Neither `ifconfig` nor `ip` commands are available, listing network interfaces is likely to fail
/bin/sh: 1: ip: not found

/opt/ros/eloquent/lib/python3.6/site-packages/ros2doctor/api/format.py: 76: UserWarning: Fail to call NetworkReport class functions.

  PLATFORM INFORMATION
system           : Linux
platform info    : Linux-4.15.0-72-generic-x86_64-with-Ubuntu-18.04-bionic
release          : 4.15.0-72-generic
processor        : x86_64

  RMW MIDDLEWARE
middleware name    : rmw_fastrtps_cpp

  ROS 2 INFORMATION
distribution name      : eloquent
distribution type      : ros2
distribution status    : active
release platforms      : {'ubuntu': ['bionic']}

  TOPIC LIST
topic               : none
publisher count     : 0
subscriber count    : 0
```

Having the error up top is quite clear I think and at least the other info is shown.