ros2/ros2cli #427: add support for additional build-types, e.g., ros2 pkg create --build-type anewbuildtype

wayneparrott, 2020-01-02:
This enhancement request is to enable the creation of packages for build-types other than c/c++ and python, i.e.,`ros2 pkg create --build-type mynewbuildtype`

Specifically I'm proposing the addition of an extension point for additional build-types. Something like this:

```python
entry_points={
        'ros2cli.extension_point': [
            'ros2pkg.build_type = ros2pkg.build_type:BuildTypeExtension',
        ],
```

As an experiment I implemented this extension point and it was quite simple to work into  ros2pkg.verb.command.py with minimal changes, no refactoring of existing build-types required. I've been using my implementation for creating javascript/[rclnodejs](https://github.com/RobotWebTools/rclnodejs) projects. I would be glad to share a PR for your consideration. If included in ros2cli, we (the rclnodejs team) will release a new build type for creating nodejs projects that properly work with `ros2 run` and `ros2 launch` commands.  

Thoughts?

dirk-thomas, 2020-01-02:
That sounds great. Please create a PR with that proposal.

wayneparrott, 2020-01-27:
@ivanpauno I'm in the process of prep'ing a PR for this enhancement. I'm blocked with a dev environment issue running the ros2pkg/test_cli.py tests. Once I can run those tests successfully I'll submit the PR for review.

ivanpauno, 2020-01-28:
> @ivanpauno I'm in the process of prep'ing a PR for this enhancement. I'm blocked with a dev environment issue running the ros2pkg/test_cli.py tests. Once I can run those tests successfully I'll submit the PR for review.

Sounds great, let us know if you have any question.