ros2/ros2cli #430: Cannot echo topic as it contains more than one type (but the type naming seems the same)

TSC21, 2020-01-06:
## Bug report

I am sure that this is not a problem with the `ros2cli`, since the binding still fails, but it's kinda awkward what's happening in terms of the `topic echo` part. I am not certain what's causing this, but what I am seeing is, theoretically, the same type name being considered different.

The full description of the problem and reproduction of the steps can be found under https://github.com/eProsima/Fast-RTPS/issues/829. To add: the fact that the type name on the agent side was changed on both Publisher and Subscribers using the Fast-RTPS `TopicDataType` API:

```c++
    myType.setName("@(package)/msg/@(topic)");
```
In order to match the type naming that the ROS2 Pub/Subs expect.

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Dashing, from binaries
- Version or commit hash:
  - `ros2cli` `0.7.9-1bionic.20191212.194301`
- DDS implementation:
  - `rmw_fastrtps`
- Client library (if applicable):
  - `rclcpp`

#### Expected behavior
I expected the topic to have only data of the same type and `ros2 topic echo` actually output the data stream.

#### Actual behavior
https://github.com/eProsima/Fast-RTPS/issues/829#issuecomment-570922720:

```sh
$ ros2 topic echo /SensorCombined_PubSubTopic 
Cannot echo topic '/SensorCombined_PubSubTopic', as it contains more than one type: [px4_msgs/msg/SensorCombined, px4_msgs/msg/SensorCombined]
```

#### Additional information

All the information can be found under https://github.com/eProsima/Fast-RTPS/issues/829.

@dirk-thomas thanks in advance for your time and patience.

thomasgubler, 2020-01-08:
@tfoote @dirk-thomas FYI this issue is blocking integration of ROS2 into PX4. We might have missed something in our setup and would be very thankful for your input.

TSC21, 2020-01-10:
This was solved in https://github.com/PX4/Firmware/pull/13907.

closed 2020-01-10 by TSC21

catarinam93, 2025-03-21:
Não perceboi nada.