ros2/ros2cli #440: Unable to run ros2param tests

pbaughman, 2020-01-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 -->
- Installation type:
  - Source
- Version or commit hash:
  - d273021ac4b5a06f8aa7b6611a91bad686960374
- DDS implementation:
  - N/A
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
docker pull osrf/ros2:nightly
docker run --rm -it -v ~/ros2cli:/ros2cli osrf/ros2:nightly

# Inside the docker container:
cd ros2cli
colcon build
colcon test --packages-select ros2param
colcon test-result --verbose
root@e28c930b6173:/ros2cli# colcon test-result --verbose
build/ros2param/pytest.xml: 23 tests, 0 errors, 4 failures, 0 skipped
- ros2param.test.test_verb_dump.TestVerbDump test_verb_dump (test/test_verb_dump.py:101)
  <<< failure message
    AttributeError: 'Namespace' object has no attribute ' verb'
  >>>
- ros2param.test.test_verb_dump.TestVerbDump test_verb_dump_invalid_node (test/test_verb_dump.py:89)
  <<< failure message
    AttributeError: 'Namespace' object has no attribute ' verb'
  >>>
- ros2param.test.test_verb_dump.TestVerbDump test_verb_dump_invalid_path (test/test_verb_dump.py:96)
  <<< failure message
    AttributeError: 'Namespace' object has no attribute ' verb'
  >>>
- ros2param.test.test_verb_dump.TestVerbDump test_verb_dump_print (test/test_verb_dump.py:110)
  <<< failure message
    AttributeError: 'Namespace' object has no attribute ' verb'
  >>>
```

#### Expected behavior
Tests should run fine - this is 'master' and I'm using the ROS2 docker image

#### Actual behavior
Strange errors.  If I cat the stdout_stderr log from the test:

```
cat log/latest_test/ros2param/stdout_stderr.log
. . .
[9.847s] =================================== FAILURES ===================================
[9.847s] _________________________ TestVerbDump.test_verb_dump __________________________
[9.848s] test/test_verb_dump.py:104: in test_verb_dump
[9.849s]     assert cli.main(
[9.849s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/cli.py:43: in main
[9.850s]     required=False, argv=argv)
[9.851s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:242: in add_subparsers_on_demand
[9.851s]     **kwargs)
[9.852s] ros2param/command/param.py:27: in add_arguments
[9.852s]     parser, cli_name, '_verb', 'ros2param.verb', required=False)
[9.852s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:205: in add_subparsers_on_demand
[9.852s]     name = getattr(known_args, subparser.dest)
[9.853s] E   AttributeError: 'Namespace' object has no attribute ' verb'
[9.853s] ___________________ TestVerbDump.test_verb_dump_invalid_node ___________________
[9.853s] test/test_verb_dump.py:91: in test_verb_dump_invalid_node
[9.853s]     assert cli.main(
[9.853s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/cli.py:43: in main
[9.853s]     required=False, argv=argv)
[9.854s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:242: in add_subparsers_on_demand
[9.854s]     **kwargs)
[9.854s] ros2param/command/param.py:27: in add_arguments
[9.854s]     parser, cli_name, '_verb', 'ros2param.verb', required=False)
[9.854s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:205: in add_subparsers_on_demand
[9.854s]     name = getattr(known_args, subparser.dest)
[9.855s] E   AttributeError: 'Namespace' object has no attribute ' verb'
[9.855s] ___________________ TestVerbDump.test_verb_dump_invalid_path ___________________
[9.855s] test/test_verb_dump.py:98: in test_verb_dump_invalid_path
[9.855s]     assert cli.main(
[9.855s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/cli.py:43: in main
[9.856s]     required=False, argv=argv)
[9.856s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:242: in add_subparsers_on_demand
[9.856s]     **kwargs)
[9.856s] ros2param/command/param.py:27: in add_arguments
[9.856s]     parser, cli_name, '_verb', 'ros2param.verb', required=False)
[9.857s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:205: in add_subparsers_on_demand
[9.857s]     name = getattr(known_args, subparser.dest)
[9.857s] E   AttributeError: 'Namespace' object has no attribute ' verb'
[9.857s] ______________________ TestVerbDump.test_verb_dump_print _______________________
[9.857s] test/test_verb_dump.py:113: in test_verb_dump_print
[9.857s]     assert cli.main(
[9.858s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/cli.py:43: in main
[9.858s]     required=False, argv=argv)
[9.858s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:242: in add_subparsers_on_demand
[9.858s]     **kwargs)
[9.858s] ros2param/command/param.py:27: in add_arguments
[9.859s]     parser, cli_name, '_verb', 'ros2param.verb', required=False)
[9.859s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:205: in add_subparsers_on_demand
[9.859s]     name = getattr(known_args, subparser.dest)
[9.859s] E   AttributeError: 'Namespace' object has no attribute ' verb'
```
Am I doing something wrong the way I run the tests?  I'd expect these to work out of the box

#### Additional information
If I add a few additional print statements around where it's crashing:

```
# check if a specific subparser is selected
print(known_args)
print(subparser.dest)
name = getattr(known_args, subparser.dest)
```

I can see:

```
[4.327s] _________________________ TestVerbDump.test_verb_dump __________________________
[4.328s] test/test_verb_dump.py:104: in test_verb_dump
[4.328s]     assert cli.main(
[4.329s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/cli.py:43: in main
[4.330s]     required=False, argv=argv)
[4.331s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:244: in add_subparsers_on_demand
[4.334s]     **kwargs)
[4.335s] ros2param/command/param.py:27: in add_arguments
[4.336s]     parser, cli_name, '_verb', 'ros2param.verb', required=False)
[4.336s] ../install/ros2cli/lib/python3.6/site-packages/ros2cli/command/__init__.py:207: in add_subparsers_on_demand
[4.337s]     name = getattr(known_args, subparser.dest)
[4.338s] E   AttributeError: 'Namespace' object has no attribute ' verb'
[4.338s] ----------------------------- Captured stdout call -----------------------------
[4.339s] Namespace(**{' command': 'param'})
[4.339s]  command
[4.340s] Namespace(**{' command': None})
[4.340s]  verb
```

dirk-thomas, 2020-01-25:
Same result in the nightly CI: https://ci.ros2.org/view/nightly/job/nightly_linux_debug/1438/testReport/junit/ros2param.test.test_verb_dump/

closed 2020-01-25 by dirk-thomas

pbaughman, 2020-01-25:
Ah, just bad luck then.  Thanks Dirk!

PR #441 (2020-01-25): fix ros2param tests
Fixes #440 which was a regression of #436.

Same fix as ros2/sros2#175.

CI build only testing `ros2param`: [image]