ros2/ros2cli #456: ros2node for rmw_cyclonedds_cpp: test_cli::test_list_node_count/test_list_all_nodes_count misses launch_ros

brawner, 2020-02-25:
This is a regular failure on nightly builds, but for some reason the node count is off by one only for `rmw_cyclonedds_cpp`. Inspecting the output of ros2 node list, it reports:

/_hidden_complex_node
/_ros2cli_daemon_1
/complex_node

The test appears to be missing `launch_ros`

https://ci.ros2.org/view/nightly/job/nightly_win_extra_rmw_rel/673/testReport/ros2node.test/test_cli/test_cli/
https://ci.ros2.org/view/nightly/job/nightly_linux_extra_rmw_release/619/testReport/ros2node.test/test_cli/test_cli/

rotu, 2020-03-19:
@brawner please clarify what you mean that it "misses" launch_ros. What is the issue?

brawner, 2020-03-19:
During the above linked failures, the expected number of nodes doesn't match what it actually finds. When I was digging into this, the launch_ros node is running, but it's not found during [this test](https://github.com/ros2/ros2cli/blob/beadc01db22d8f77c281e86a4e2c2634db0698a5/ros2node/test/test_cli.py#L142)

```
Traceback (most recent call last):
  File "/home/jenkins-agent/workspace/nightly_linux_extra_rmw_release/ws/install/launch_testing/lib/python3.6/site-packages/launch_testing/markers.py", line 48, in _wrapper
    return func(self, *args, **kwargs)
  File "/home/jenkins-agent/workspace/nightly_linux_extra_rmw_release/ws/src/ros2/ros2cli/ros2node/test/test_cli.py", line 153, in test_list_all_nodes_count
    assert int(output_lines[0]) == 4
AssertionError: assert 3 == 4
 +  where 3 = int('3')
```

It still is happening in recent tests, you can search the consoleText:
`FAIL: test_list_all_nodes_count[rmw_cyclonedds_cpp] (test_cli.TestROS2NodeCLI)`
https://ci.ros2.org/view/nightly/job/nightly_linux_extra_rmw_release/653/consoleText

rotu, 2020-03-19:
Ah... Seems to be caused by https://github.com/ros2/launch_ros/pull/128

brawner, 2020-03-19:
This failure has been ongoing for awhile now and still occurs after that PR was merged, but I'm also not familiar enough with this everything involved to suggest what's causing the issue.

rotu, 2020-03-19:
Ya. It looks like the nodes aren't being "missed" - they just don't exist anymore. And it seems to have been *caused* at least in part by that PR. So it's a naughty test, not a misbehavior of `ros2 node list -a -c`.

brawner, 2020-03-19:
Now the behavior is more consistent among the different DDS implementations. Maybe the test just needs to be updated? It fails now with connext, fastrtps, cyclonedds

Some more examples
https://ci.ros2.org/view/nightly/job/nightly_linux_release/1486/testReport/ros2service.src.ros2.ros2cli.ros2service.test/test_cli/test_cli/
https://ci.ros2.org/view/nightly/job/nightly_win_extra_rmw_rel/703/testReport/ros2node.src.ros2.ros2cli.ros2node.test/test_cli/test_cli/

clalancette, 2024-12-27:
This is an old issue, and I don't think that we are having this exact problem anymore in CI, so I'm going to close this.  Feel free to reopen if you still think this is a problem.

closed 2024-12-27 by clalancette