ros2/ros2cli #457: Test failures from formatting of several test_cli.py tests with rmw_cyclonedds_cpp

brawner, 2020-02-25:
This is consistent issue across multiple packages, but rmw_cyclonedds_cpp runs of test_cli.py produce slightly different outputs than what the tests expect. This fails on multiple platforms.

Affected packages:
 [ros2topic](https://ci.ros2.org/view/nightly/job/nightly_linux_extra_rmw_release/619/testReport/ros2topic.test/test_cli/test_cli/)
[ros2lifecycle](https://ci.ros2.org/view/nightly/job/nightly_linux_extra_rmw_release/619/testReport/ros2lifecycle.test/test_cli/test_cli/)
[ros2action](https://ci.ros2.org/view/nightly/job/nightly_linux_extra_rmw_release/619/testReport/ros2action.test/test_cli/test_cli/)
[ros2service](https://ci.ros2.org/view/nightly/job/nightly_linux_extra_rmw_release/619/testReport/ros2service.test/test_cli/test_cli/)

rotu, 2020-03-19:
These inter-RMW failures seem to have a similar cause stem from this issue:
```
[INFO] [simple_lifecycle_node-31]: process started with pid [12951]
...
[simple_lifecycle_node-31] 1582493368.245026 [108] dq.builtin: plist(vendor 1.15): parameter length 39 mod 4 != 0
[simple_lifecycle_node-31] 1582493368.245040 [108] dq.builtin: SPDP (vendor 1.15): invalid qos/parameters
[simple_lifecycle_node-31] 1582493368.245044 [108] dq.builtin: plist(vendor 1.15): parameter length 39 mod 4 != 0
[simple_lifecycle_node-31] 1582493368.245047 [108] dq.builtin: SPDP (vendor 1.15): invalid qos/parameters
[simple_lifecycle_node-31] 1582493368.245116 [108] dq.builtin: plist(vendor 1.15): parameter length 39 mod 4 != 0
```

rotu, 2020-03-19:
Okay. this seems to be a defect in FastRTPS and fixed in https://github.com/eProsima/Fast-RTPS/pull/1047

Recommend closing this issue

brawner, 2020-03-19:
This does seem to be resolved, closing.

closed 2020-03-19 by brawner