ros2/ros2cli #475: Create meta-package with ros2cli common extensions

ivanpauno, 2020-04-02:
## Feature request

#### Feature description

Create a `ros2cli_common_extensions` metapackage.
It will make easier for example to build all the `ros2cli` extensions.

#### Implementation considerations

It should probably include `ros2launch` and `sros2`, so it should probably be in a separated repository.

sunbo57123, 2020-06-08:
hello, 
I'm trying to resolve this issue, after reading some documents, I want to know if this could be a repo that seems like [ros/metapackage](https://github.com/ros/metapackages/).

ivanpauno, 2020-06-08:
> hello,
> I'm trying to resolve this issue, after reading some documents, I want to know if this could be a repo that seems like [ros/metapackage](https://github.com/ros/metapackages/).
> If there are some more docs or context of this issue could refer to?

That repository is not related to this.

In ROS 2, a metapackage is an "empty" package that collects other packages (so you can then easily build all of them, or install their debians).
See for example: https://github.com/ros2/rosbag2/tree/master/rosbag2.

To start, I would open a PR adding the new metapackage in this repo.
The package can be named `ros2cli_common_extensions`.

sunbo57123, 2020-06-09:
Much thanks for your response! 
There are still 2 problems that confuse me:
- The content of this repo should contain `ros2launch` and `sros2`,is this designed for downloading all the cli commands(ros2cli+sros2+ros2launch) through metapackage? Is `ros2launch` here is repo **[ros2/launch_ros/ros2launch/](https://github.com/ros2/launch_ros/tree/master/ros2launch)** or **[ros2/launch](https://github.com/ros2/launch)**?
- In c++ projects, the configure file is CMakeList, and in python project, should this part be handled through the setup file? 
maybe  https://github.com/PR2/pr2_robot/tree/kinetic-devel/pr2_camera_synchronizer is an example for metapackage in python project?

ivanpauno, 2020-06-09:
It can be an empty c++ project, it doesn't matter if it collects both python and c++ packages.
The only things that matters is that the manifest should collect all the desired dependencies (e.g. https://github.com/ros2/rosbag2/blob/2a939386ff2134a67e3708b8f75b0d68c6015311/rosbag2/package.xml#L12-L20).

> The content of this repo should contain ros2launch and sros2,is this designed for downloading all the cli commands(ros2cli+sros2+ros2launch) through metapackage? Is ros2launch here is repo ros2/launch_ros/ros2launch/ or ros2/launch?

`ros2launch` and `sros2` will continue being in their repositories.
The only problem with adding `ros2cli_common_extensions` here, is that it will create a circular dependency between repositories (e.g. `launch_ros` and this repo).
That creates some problems in the releasing process, so we will probably add `ros2cli_common_extensions` to a new repository instead of here.
But to start discussing what we want to include in it, you can send a PR here (we can later move it).

> In c++ projects, the configure file is CMakeList, and in python project, should this part be handled through the setup file?
maybe https://github.com/PR2/pr2_robot/tree/kinetic-devel/pr2_camera_synchronizer is an example for metapackage in python project?

That's not a meta-package. As commented above, you can create an empty `ament_cmake` project.
Creating this kind of empty packages is useful for example when building from source:
```
colcon build --packages-up-to ros2cli_common_extensions
```
will build all ros2cli tools (currently you have to write a long list of packages).
It could also be used to install all ros2cli tools from debians:
```
sudo apt install ros-<distro>-ros2cli-common-extensions
```

ivanpauno, 2020-07-16:
Done in https://github.com/ros2/ros2cli_common_extensions/pull/2.

ivanpauno, 2020-07-16:
Thanks @sunbo57123 for the implementation!

closed 2020-07-16 by ivanpauno