ros2/ros2cli #476: ros2 node list not working

jacobperron, 2020-04-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 786a5bf754cbb5667cb677e8a902879dfad1c413
- DDS implementation:
  - Fast-RTPS (1.9.x)

#### Steps to reproduce issue

Start a talker

```
ros2 run demo_nodes_cpp talker
```

List nodes with:

```
ros2 node list
```

#### Expected behavior

We see the talker in the list

```
/talker
```

#### Actual behavior

I see the following output:

```
//
```

#### Additional information

For each node in the graph, we see a line `//`.

This does not happen with CycloneDDS; I have not tried with Connext.

ivanpauno, 2020-04-08:
I have checked locally, I can't reproduce this.

`rcl` [test_get_node_names](https://github.com/ros2/rcl/blob/master/rcl/test/rcl/test_get_node_names.cpp) is passing on the [nightlies](https://ci.ros2.org/view/nightly/job/nightly_linux_release/1506/testReport/rcl/TestGetNodeNames__rmw_fastrtps_cpp/), so I would be surprised if this is failing.

jacobperron, 2020-04-08:
Okay. I'm closing assuming I've done something to my machine to cause this.

closed 2020-04-08 by jacobperron