ros2/ros2cli #480: ros2param test failures (cross-platform)

jacobperron, 2020-04-09:
## Bug report

**Required Info:**

- Operating System:
  - All
- Installation type:
  - source
- Version or commit hash:
  - c9e9e2f9e932c8775d59aff3a5ac833f918afdbd
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

See nightly failures:

- https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/1836/testReport/junit/ros2param.src.ros2.ros2cli.ros2param.test.test_verb_dump/TestVerbDump/test_verb_dump/
- https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/1836/testReport/junit/ros2param.src.ros2.ros2cli.ros2param.test.test_verb_dump/TestVerbDump/test_verb_dump_invalid_path/
- https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/1836/testReport/junit/ros2param.src.ros2.ros2cli.ros2param.test.test_verb_dump/TestVerbDump/test_verb_dump_print/

#### Expected behavior

Tests pass

#### Actual behavior

Tests fail

#### Additional information
Roughly, it looks like the failures started on April 5th ([based on the nightly Linux repeated build](https://ci.ros2.org/view/nightly/job/nightly_linux_repeated/1832/)).

jacobperron, 2020-04-09:
It looks like the failures are because not enough time is given for discover to happen between the test node and the CLI node (ie. the daemon).

If I switch the RMW implementation to `rmw_connext_cpp`, then I am able to reproduce reliably (I think because it takes longer for Connext to initialize).

jacobperron, 2020-04-10:
#481 isn't the fix we were looking for. Now we can see a timeout happening during the newly added wait for discovery: https://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_repeated/1158/testReport/ros2param.src.ros2.ros2cli.ros2param.test.test_verb_dump/TestVerbDump/test_verb_dump/

jacobperron, 2020-04-10:
I can reproduce the timeouts locally by first starting the daemon with an RMW implementation that is different than the one used by the tests:

```
RMW_IMPLEMENTATION=rmw_connext_cpp ros2 daemon start
```

And then  running the tests
```
colcon test --packages-select ros2param
```

hidmic, 2020-04-10:
Aha! That's why most `ros2cli` _launch_ tests systematically stop that daemon e.g. https://github.com/ros2/ros2cli/blob/master/ros2topic/test/test_cli.py#L116.

jacobperron, 2020-04-10:
@hidmic I noticed that #389 doesn't touch `test_verb_dump.py`, so I started converting that set of tests to launch tests, independent of #389.

commit 2d060099 (2020-04-09): [ros2param] Wait for discovery before running tests

Fixes #480.

Catch expected exceptions from rclpy (or transitively as xmlrpc.client.Fault) while we wait for discovery in the test setup.
PR #481 (2020-04-10): [ros2param] Wait for discovery before running tests
Fixes #480.

Catch expected exceptions from rclpy (or transitively as xmlrpc.client.Fault) while we wait for discovery in the test setup.
commit 8e6e051a (2020-04-10): [ros2param] Wait for discovery before running tests (#481)

Fixes #480.

Catch expected exceptions from rclpy (or transitively as xmlrpc.client.Fault) while we wait for discovery in the test setup.
closed 2020-04-18 by jacobperron