ros2/ros2cli #483: rosed and roscd equivalent in ros2cli

Briancbn, 2020-04-12:
## Feature request

#### Feature description
A `ros2edit` and `ros2cd` package that can easily navigate and edit the workspace source folder, like how `rosed` and `roscd` can do in ROS 1, which was a huge convenience for me when there was no gui.

#### Implementation considerations
Most of the implementation will be similar to `ros2run`, but there will be a need to locate package path inside the workspace instead of the index space, I find that `colcon list` can return the information, but might be too slow and colcon have their own common extensions(open for suggestions).

I think `ROS_WORKSPACE` will also need to be specified, but without it we can look for packages from the current path.

I can try and write up a pr, but please let me know if it is needed or you have any suggestions.

dirk-thomas, 2020-04-14:
Please see #75 which describes why a `roscd` command isn't available and that `colcon_cd` (colcon/colcon-cd#1) provide an alternative.

For `rosed` the challenges are very similar than for `roscd`. I am not sure if it makes sense to implement something like this as a colcon extension (similar to `colcon_cd`).

Briancbn, 2020-04-14:
Yes it makes much more sense actually. I think I will try to do it as a colcon extension.

hidmic, 2020-04-16:
@dirk-thomas @Briancbn would it make sense to close this issue / transfer it to a `colcon` repo?

dirk-thomas, 2020-04-16:
I will go ahead with closing the ticket here.

@Briancbn Please feel free to contribute to colcon. If the addition is in form of a new repo please create it under your own GitHub user and when it is ready we can move it into the `colcon` org unit (as per [recommendation](https://colcon.readthedocs.io/en/released/developer/contribution.html#new-packages-extensions)).

closed 2020-04-16 by dirk-thomas