ros2/ros2cli #492: ros2topic test_echo_pub can't handle new QoS mismatch warnings

eboasson, 2020-04-20:
**Required Info:**

- Operating System:
  - macOS 10.15.4 (should be platform independent)
- Installation type:
  - from source
- Version or commit hash:
  - 75e0ea1f5da57321dd3d6e78e842a4f8fe68a3d6 (master as of 19-4-2020)
- DDS implementation:
  - rmw_cyclonedds_cpp

A bit of experimentation shows that it is the new warning when incompatible QoS's are detected that throws off the test. They depend on a feature supported by Cyclone DDS but not yet by Fast-RTPS.

#### Steps to reproduce issue
```
(cd src/ros2/ros2cli/ros2topic/test && pytest test_echo_pub.py)
```

#### Expected behavior

Test passes

#### Actual behavior

Test fails with

```
======================================================================
FAIL: test_echo_basic (test_echo_pub.TestROS2TopicEchoPub) (topic='/clitest/topic/echo_incompatible_qos', provide_qos=True, compatible_qos=False)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/erik/ros2_ws/src/ros2/ros2cli/ros2topic/test/test_echo_pub.py", line 250, in test_echo_basic
    assert not command.output, (
AssertionError: Echo CLI should not have received anything with incompatible QoS
assert not 'Incompatible QoS Policy detected: RELIABILITY_QOS_POLICY\n'
 +  where 'Incompatible QoS Policy detected: RELIABILITY_QOS_POLICY\n' = <launch_testing.tools.process.ProcessProxy object at 0x1090be7d0>.output
```

mm318, 2020-04-20:
> * DDS implementation:
>   * rmw_cyclonedds_cpp

> They depend on a feature supported by Cyclone DDS but not yet by Fast-RTPS.

So is this a problem with `rmw_cyclonedds_cpp` or `rmw_fastrtps_cpp`?

I wasn't able to reproduce the test failure with just:
> `cd src/ros2/ros2cli/ros2topic/test && pytest test_echo_pub.py`

Is there some environment variable that needs to be set?

dirk-thomas, 2020-04-21:
> So is this a problem with `rmw_cyclonedds_cpp` or `rmw_fastrtps_cpp`?

As the above description clearly mentions:

    DDS implementation:
        rmw_cyclonedds_cpp

> Is there some environment variable that needs to be set?

To choose a non-default RMW you have to set e.g.:

    RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

closed 2020-05-04 by ivanpauno