ros2/ros2cli #502: ROS2 daemon terminates silently

cnboonhan94, 2020-05-04:
**Context:** 
I am currently using the Server-Client discovery mechanism from FastRTPS, as documented [here](https://fast-rtps.docs.eprosima.com/en/latest/advanced.html#server-client-discovery), for a ROS2 network. This "enterprise" network connects multiple devices, both wired and wireless, via Cisco Catalyst 3850 switches and Cisco Access Points. The wired and wireless devices are separated into two VLANs. For this example, only 3 devices are live: a Discovery Server, Talker and Listener.

**Issue:**
Initial runs with the 3 device setup ( Discovery Server, Talker publishing, Listener subscribing ) work well. However, after a certain amount of time ( ranging from 1 hour to more ), the `ros2_daemon` process on the Discovery Server will eventually terminate. The consequence is that any new subscriptions from Listener to Talker will not work. The `ros2_daemon` process does not automatically restart, thus crippling the discovery mechanism.

I have not been able to replicate this issue on a simpler ( flat ) wired switched network with single ASUS router the Discovery Server, Talker and Listener, within a similar amount of time.

**Follow up:**
I am currently using a `cron` job that checks for `ros2_daemon` and restarts it if it is down. 

However, I would also like to know what sort of actions I can take to search for any relevant information that might reveal information about why the process terminates. 

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - ROS2 Eloquent, installed from binaries
- Version or commit hash:
  - Latest Eloquent 
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

For all devices, we expect RO2 Eloquent to be sourced: `source /opt/ros/eloquent/setup.bash`

** Network Hardware **
It is not possible to document the specific configuration of the network. Some details are specified in the primer above, but let me know if there is any more information I could provide. Again, I was not able to replicate this one a simpler network, within the time I had.

**On Discovery Server**
1. add the following as `$HOME/server_config.xml`
```
<?xml version="1.0" encoding="utf-8" ?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
  <profiles>
    <participant profile_name="UDP SERVER" is_default_profile="true">
      <rtps>
        <prefix>72.61.73.70.66.61.72.6d.74.65.73.74</prefix>
        <builtin>
          <discovery_config>
            <discoveryProtocol>SERVER</discoveryProtocol>
          </discovery_config>
          <metatrafficUnicastLocatorList>
            <locator>
              <udpv4>
                <address>10.233.0.2</address>
                <port>56542</port>
              </udpv4>
            </locator>
          </metatrafficUnicastLocatorList>
        </builtin>
      </rtps>
    </participant>
  </profiles>
</dds>
```
2. `export FASTRTPS_DEFAULT_PROFILES_FILE=$HOME/server_config.xml`
3. `ros2 daemon start`

**On Talker**
1. Add the following as `$HOME/client_config.xml`
```
<?xml version="1.0" encoding="utf-8"?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
[... 294 lines cut ...]
Apr 30 21:49:33 server-room dbus-daemon[1695]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Apr 30 21:49:33 server-room systemd[1]: Started Network Manager Script Dispatcher Service.
Apr 30 21:49:33 server-room nm-dispatcher: req:1 'dhcp4-change' [enp0s20f0u11]: new request (2 scripts)
Apr 30 21:49:33 server-room nm-dispatcher: req:1 'dhcp4-change' [enp0s20f0u11]: start running ordered scripts...
Apr 30 22:04:14 server-room systemd[1]: Started Run anacron jobs.
Apr 30 22:04:14 server-room anacron[35703]: Anacron 2.3 started on 2020-04-30
Apr 30 22:04:14 server-room anacron[35703]: Normal exit (0 jobs run)
Apr 30 22:16:55 server-room dhclient[5093]: DHCPREQUEST of 192.168.42.205 on enp0s20f0u11 to 192.168.42.129 port 67 (xid=0x3f170e62)
Apr 30 22:16:55 server-room dhclient[5093]: DHCPACK of 192.168.42.205 from 192.168.42.129
Apr 30 22:16:55 server-room NetworkManager[1717]: <info>  [1588256215.0820] dhcp4 (enp0s20f0u11):   address 192.168.42.205
Apr 30 22:16:55 server-room NetworkManager[1717]: <info>  [1588256215.0821] dhcp4 (enp0s20f0u11):   plen 24 (255.255.255.0)
Apr 30 22:16:55 server-room NetworkManager[1717]: <info>  [1588256215.0821] dhcp4 (enp0s20f0u11):   gateway 192.168.42.129
Apr 30 22:16:55 server-room NetworkManager[1717]: <info>  [1588256215.0822] dhcp4 (enp0s20f0u11):   lease time 3600
Apr 30 22:16:55 server-room NetworkManager[1717]: <info>  [1588256215.0822] dhcp4 (enp0s20f0u11):   hostname 'server-room'
Apr 30 22:16:55 server-room NetworkManager[1717]: <info>  [1588256215.0822] dhcp4 (enp0s20f0u11):   nameserver '192.168.42.129'
Apr 30 22:16:55 server-room NetworkManager[1717]: <info>  [1588256215.0823] dhcp4 (enp0s20f0u11): state changed bound -> bound
Apr 30 22:16:55 server-room dbus-daemon[1695]: [system] Activating via systemd: service name='org.freedesktop.nm_dispatcher' unit='dbus-org.freedesktop.nm-dispatcher.service' requested by ':1.13' (uid=0 pid=1717 comm="/usr/sbin/NetworkManager --no-daemon " label="unconfined")
Apr 30 22:16:55 server-room dhclient[5093]: bound to 192.168.42.205 -- renewal in 1347 seconds.
Apr 30 22:16:55 server-room systemd[1]: Starting Network Manager Script Dispatcher Service...
Apr 30 22:16:55 server-room dbus-daemon[1695]: [system] Successfully activated service 'org.freedesktop.nm_dispatcher'
Apr 30 22:16:55 server-room systemd[1]: Started Network Manager Script Dispatcher Service.
Apr 30 22:16:55 server-room nm-dispatcher: req:1 'dhcp4-change' [enp0s20f0u11]: new request (2 scripts)
```

</details>

codebot, 2020-05-05:
Some additional data:
It appears to be related to a timeout triggered by https://github.com/ros2/ros2cli/blob/master/ros2cli/ros2cli/daemon/__init__.py#L43-L45    by adjusting that timeout to something very short (for example, 10 seconds), the behavior reliably occurs, without needing to wait for two hours.

The console prints the expected message on https://github.com/ros2/ros2cli/blob/master/ros2cli/ros2cli/daemon/__init__.py#L111   whether on the "stock" 2-hour setup, or the modified much-shorter timeout, so we believe the same failure mode is triggered in both cases.

This issue is currently looking like an unintended interaction between the daemon's timeout and the FastRTPS Discovery Service. In particular, the daemon can silently be forked when running `ros2 topic list` , or explicitly launched via `ros2 daemon start`. It will use the Discovery Service Server configuration if `FASTRTPS_DEFAULT_PROFILES_FILE` is in the environment, pointing to a configuration XML that configures it as such. At that point, the Discovery Service Server is now part of the daemon, which is now running its countdown timer. If this is running on an otherwise "quiet" machine (ironically in an attempt to "protect" the stability of the critical Discovery Service), there will be no localhost queries to the daemon, and after the timeout, :boom:   :rofl:  all discovery clients on the network are toast.

For now, we will try explicitly running the Discovery Service Server configuration with a known "talker" node so that it's not running a silent background countdown timer :rofl:  in the future perhaps we can evolve some tools to help set up and debug this type of "top-level" networking configuration in cases where multicast discovery is not an option.

ivanpauno, 2020-05-05:
Oh, I didn't read the description thoroughly before.
Using the daemon (with a timeout) as the discovery server doesn't sound like a great idea.

Maybe, an argument to launch the daemon without a timeout can be added (that argument could be the timeout itself, where negative numbers would mean run "forever").

There's also a [BACKUP](https://github.com/eProsima/Discovery-Server#discoverysettings) server, which can be restored.

AFAIU, you can also create multiple discovery servers, that probably adds reliability.

----

The only actionable item I see for `ros2cli`, is adding an option to avoid the timeout.
I don't know if you have something else in mind @codebot @cnboonhan94 .

cnboonhan94, 2020-05-06:
I think that would be sufficient to guard against a silent timeout. :+1:

codebot, 2020-05-06:
Unless I'm misunderstanding something, I don't believe it is currently possible to set the timeout when running `ros2 daemon start`. It would be nice to have that configurable, in addition to a negative (or zero) number to indicate "indefinite"

ivanpauno, 2020-05-06:
> Unless I'm misunderstanding something, I don't believe it is currently possible to set the timeout when running ros2 daemon start. It would be nice to have that configurable, in addition to a negative (or zero) number to indicate "indefinite"

Yes, currently it isn't, and AFAIS that's the only improvement it can be done here.
@cnboonhan94 @codebot are you planning to propose a PR adding that? (it won't probably be merged until Foxy is released).

dirk-thomas, 2020-05-06:
> are you planning to propose a PR adding that? (it won't probably be merged until Foxy is released).

Adding a trivial command line option to make the timeout configurable can be merged as soon as a pull request is available and had been reviewed.

MiguelBarro, 2020-05-08:
We have been studying the issue and conclude it has nothing to do with rmw implementation. We took the following steps:

1. We simplify the scenario. At first we used docker containers to simulate the machines of @cnboonhan94 scenario. We were able to reproduce the issue but after some testing we decided to simplify it. We end up using a single container with only one daemon (set as a server), a talker and a listener. The issue takes place also in the simplified scenario (it takes half an hour for the daemon to disappear).  

2. Then we repeat the simplified scenario using Fast-RTPS as rmw and simple discovery protocol. The daemon disappears also when we use simple discovery (is an ordinary participant), thus, it's not a client-server discovery issue.

3. Finally we repeat the simplified scenario using Cyclone as rmw. The daemon disappears also, thus, we conclude it's not a Fast-RTPS issue. We conclude the issue stems from above the rmw layer.

# Step-by-step testing

We produced a focal docker from the last ros2 master **.repos** available. To keep the docker image manageable we only install the ros2 packages required for the testing. The docker can be downloaded from DockerHub as:

```bash
$ docker pull miguelbarroeprosima/ros2:ros2_demo_focal 
```

we are going to profit from DDS domain isolation capabilities to run the three tests in a single docker.

## Test A - fastrtps client-server 

We modified the xml files in order to use localhost as server IP address:

### client_config.xml 

```xml

<?xml version="1.0" encoding="utf-8"?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
  <profiles>
    <participant profile_name="UDP CLIENT" is_default_profile="true">
      <rtps>
        <builtin>
          <discovery_config>
            <discoveryProtocol>CLIENT</discoveryProtocol>
            <discoveryServersList>
              <RemoteServer prefix="72.61.73.70.66.61.72.6d.74.65.73.74">
                <metatrafficUnicastLocatorList>
                  <locator>
                    <udpv4>
                      <address>127.0.0.1</address>
                      <port>56542</port>
                    </udpv4>
                  </locator>
                </metatrafficUnicastLocatorList>
              </RemoteServer>
            </discoveryServersList>
          </discovery_config>
        </builtin>
      </rtps>
    </participant>
  </profiles>
</dds>
```

### server_config.xml

```xml
<?xml version="1.0" encoding="utf-8" ?>
<dds xmlns="http://www.eprosima.com/XMLSchemas/fastRTPS_Profiles">
  <profiles>
    <participant profile_name="UDP SERVER" is_default_profile="true">
      <rtps>
        <prefix>72.61.73.70.66.61.72.6d.74.65.73.74</prefix>
        <builtin>
          <discovery_config>
            <discoveryProtocol>SERVER</discoveryProtocol>
          </discovery_config>
          <metatrafficUnicastLocatorList>
            <locator>
              <udpv4>
[... 79 lines cut ...]
In this case we need to modify the domain id in order to avoid collisions with the Fast-RTPS simple discovery.

```bash
# ROS_DOMAIN_ID=1 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 daemon start &
# ROS_DOMAIN_ID=1 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp listener & 
# ROS_DOMAIN_ID=1 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ros2 run demo_nodes_cpp talker 
```

Finally we list all processes to identify the daemon pids:

```bash
$ docker container top ros2_demo_focal
```

we get something like this:

| PID               | USER             |  TIME              | COMMAND           |
|:-----------------:|:----------------:|:------------------:|:-----------------:|
| ...                                                                           |
| 25497             | root             |  0:00              | {_ros2_daemon} ...|
| 25557             | root             |  0:00              | {_ros2_daemon} ...|
| 25629             | root             |  0:00              | {_ros2_daemon} ...|
| ...                                                                           |

after some time (about 2 hours) all three processes disappear.

ivanpauno, 2020-05-08:
@MiguelBarro the daemon disappearing after two hours is not related with the `rmw` implementation, that for sure.

It was designed in that way, so it doesn't run forever if not used.
Adding a way to configure/disable the timeout is a possible enhancement to solve the issue for this particular use case.

----

What is happening, is that the daemon finishes, and as it's also being used as the discovery server the clients then cannot reconnect.
My question is: if the discovery server finishes, is expected that clients can't reconnect even when the server is relaunched under the same conditions (qos/network interface)? (which I think that's what the user tried).

If the answer to that is yes, then there's no issue, and not being able to communicate after the daemon timeouts when used as a discovery server is expected.

MiguelBarro, 2020-05-09:
@ivanpauno The reason I posted our works to exculpate the rmw (when the timeout issue was already hightlighted) is that I was unable to modify the timeout time (unlike the other daemon parameters there isn't an environment variable currently associated). I agree with @cnboonhan94 that make this setting accessible is the path to go.

Concerning your question, if the server is started again discovery is recovered. It's easy to verify it using the daemon, launch the testing scenario using three sessions:

```bash
1# FASTRTPS_DEFAULT_PROFILES_FILE=/home/jenkins-agent/workspace/server_config.xml ros2 daemon start 
2# FASTRTPS_DEFAULT_PROFILES_FILE=/home/jenkins-agent/workspace/client_config.xml ros2 run demo_nodes_cpp listener  
3# FASTRTPS_DEFAULT_PROFILES_FILE=/home/jenkins-agent/workspace/client_config.xml ros2 run demo_nodes_cpp talker 
```

now we stop the listener and the daemon (this kills the server participant):

```bash
1# ros2 daemon stop 
2# ^C 
```

if we relaunch the listener and the daemon (this recreates the server paticipant), the talker reconnects to the server and discovers the new listener:

```bash
1# FASTRTPS_DEFAULT_PROFILES_FILE=/home/jenkins-agent/workspace/server_config.xml ros2 daemon start 
2# FASTRTPS_DEFAULT_PROFILES_FILE=/home/jenkins-agent/workspace/client_config.xml ros2 run demo_nodes_cpp listener  
```

using the daemon node as server was an unfortunate choice 🙃

codebot, 2020-05-19:
We are now running the Discovery Service on a generic `demo_nodes_cpp/talker` node and just saw this issue again after 2 hours :cry: 

to be more specific, we're running this on a machine that runs no other ROS 2 nodes:
```
export FASTRTPS_DEFAULT_PROFILES_FILE=$HOME/server_config.xml
nohup ros2 run demo_nodes_cpp talker --ros-args --remap __node:=discovery_talker chatter:=discovery_chatter & &>/dev/null

export FASTRTPS_DEFAULT_PROFILES_FILE=$HOME/client_config.xml
ros2 daemon stop
ros2 daemon start
```

Then we have many other machines that use that Discovery Service on a large WiFi network. It works great for 2 hours, but then it seems that when the daemon dies (as expected) after it times out, somehow that also locks up the Discovery Service running on the `talker` node on that machine. Can we disable the use of the daemon on that `talker` node that is running the Discovery Service?

codebot, 2020-05-19:
:laughing: :rofl:  I can't reliably reproduce it with shorter timeout times (for example, 60 seconds). It's maddening.... will update as I get more data.

closed 2021-09-13 by cnboonhan