ros2/ros2cli #503: ros2cli daemon test hanging with Connext

jacobperron, 2020-05-04:
## Bug report

The Foxy CI build with Connext-only is timing out (5 hours) at what looks like `test_daemon.py`.

Example build: http://build.ros2.org/view/Fci/job/Fci__nightly-connext_ubuntu_focal_amd64/98

Note, it appears to be a flake since there has been a successful build in between timeouts:

http://build.ros2.org/view/Fci/job/Fci__nightly-connext_ubuntu_focal_amd64/97/

dirk-thomas, 2020-05-04:
The first build timeout happened on Apr 18th and since then failed 5 of 7 times. That makes PR ros2/ros2cli#493 merged the day before look suspicious.

dirk-thomas, 2020-05-05:
Please see an example build where the `test_daemon` tests fails and is being repeated: https://ci.ros2.org/job/ci_linux/10624/consoleFull#console-section-204

```
_____________________ test_get_subscriptions_info_by_topic _____________________
test/test_daemon.py:204: in test_get_subscriptions_info_by_topic
    subscriptions_info = daemon_node.get_subscriptions_info_by_topic(TEST_TOPIC_NAME)
/usr/lib/python3.8/xmlrpc/client.py:1109: in __call__
    return self.__send(self.__name, args)
/usr/lib/python3.8/xmlrpc/client.py:1450: in __request
    response = self.__transport.request(
/usr/lib/python3.8/xmlrpc/client.py:1153: in request
    return self.single_request(host, handler, request_body, verbose)
/usr/lib/python3.8/xmlrpc/client.py:1169: in single_request
    return self.parse_response(resp)
/usr/lib/python3.8/xmlrpc/client.py:1341: in parse_response
    return u.close()
/usr/lib/python3.8/xmlrpc/client.py:655: in close
    raise Fault(**self._stack[0])
E   xmlrpc.client.Fault: <Fault 1: "<class 'OverflowError'>:Total nanoseconds value is too large to store in C time point.">
```

hidmic, 2020-05-05:
Alright, https://github.com/ros2/rclpy/pull/555 and 504 fixes test failures. Curious thing is that no hangs occur if the test doesn't fail. I'll try to figure out why.

closed 2020-05-07 by hidmic