ros2/ros2cli #511: --no-daemon is not valid

dawonn-haval, 2020-05-15:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - bd79e43d1273b3055cb2b431821026b44c696832
- DDS implementation:
  - fast-rtps
- Client library (if applicable):
  N/A

#### Steps to reproduce issue
```
ros2 topic list --no_daemon
```
#### Expected behavior
Should be an acceptable flag, used by strategy.py
#### Actual behavior
```
$ ros2 topic list --no_daemon
usage: ros2 [-h] Call `ros2 <command> -h` for more detailed usage. ...
ros2: error: unrecognized arguments: --no_daemon
```
#### Additional information
Likely related to the recent commit by @hidmic. This one's easy enough, I have a pull request incomming.

hidmic, 2020-05-15:
Many `ros2` command verbs are lacking `NodeStrategy` arguments. @dawonn-haval if you can fix that up, that'd be great.

dawonn-haval, 2020-05-15:
I was just getting started! :)

PR #514 (2020-07-20): Enable --no-daemon flag for some cli tools
fixes #511
added daemon support to the other topic tools, except pub.
commit acb13d98 (2020-07-20): Enable --no-daemon flag for some cli tools (#514)

Fixes #511
closed 2020-07-20 by hidmic