ros2/ros2cli #515: [ros2 topic delay] use_sim_time

doisyg, 2020-05-18:
`ros2 topic delay` seems to use the system clock instead of the topic `/clock` when using a simulation (and I guess playing a bag). 
```
$ ros2 topic delay odom
average delay: 1589796651.954
	min: 1589796651.949s max: 1589796651.958s std dev: 0.00260s window: 28

```
How to set `use_sim_time` to true for `ros2 topic` ? Or what should be the appropriate way to work with a simulation ?

hidmic, 2020-05-18:
Thanks for reporting! It is true there's currently no way to change `use_sim_time` value for `ros2cli` node. Off the top of my head, [`DirectNode`-specific arguments](https://github.com/ros2/ros2cli/blob/b622109d96e16506f1cd31cc9a4664c7df7332ec/ros2cli/ros2cli/node/direct.py#L84) could include a `-s` / `--use-sim-time` flag.

hyunseok-yang, 2021-01-06:
Hi, anyone solved this question?

clalancette, 2021-01-06:
> Hi, anyone solved this question?

It is still an open feature request, as far as I know.  If you'd like to jump in and try to implement it, that would be welcome.

fujitatomoya, 2021-01-07:
@doisyg @hyunseok-yang 

could you kindly try https://github.com/ros2/ros2cli/pull/581 to see if you can have what you want? I did make sure that ros2cli node is constructed with `use_sim_time` is true if option is enabled. it would be really appreciated.

hyunseok-yang, 2021-02-09:
@fujitatomoya Thanks. I've forgot to check your guys comment :(
I will have try the #581 code!

closed 2021-02-17 by fujitatomoya