ros2/ros2cli #519: ros2 topic echo crashes with the error when trying to echo a nonexistent topic

snehaldb90, 2020-05-28:
## Bug report

- Operating System:
  - MacOS 10.14.5 (Mojave)
- Installation type:
  - Binaries
- Version or commit hash:
  - Foxy
- DDS implementation:
  - CycloneDDS

#### Steps to reproduce issue
Running the command 'ros2 topic echo /turtle1/color_sensor' terminates with the error. Ros2 topic echo crashes when trying to echo a nonexistent topic.

#### Expected behavior
dev_ws snehaldb$ ros2 topic echo /turtle1/color_sensor
1590603774.957504 [0]     389073: using network interface en0 (udp/192.168.0.14) selected arbitrarily from: en0, utun1
r: 255
g: 255
b: 255

r: 255
g: 255
b: 255

#### Actual behavior
dev_ws snehaldb$ ros2 topic echo /turtle1/color_sensor
WARNING: topic [/turtle1/color_sensor] does not appear to be published yet
1590604507.947417 [0]     399394: using network interface en0 (udp/192.168.0.14) selected arbitrarily from: en0, utun1
Traceback (most recent call last):
  File "/Users/snehaldb/ros2_foxy/ros2-osx/bin/ros2", line 11, in <module>
    load_entry_point('ros2cli==0.9.3', 'console_scripts', 'ros2')()
  File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2cli/cli.py", line 67, in main
    rc = extension.main(parser=parser, args=args)
  File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2topic/command/topic.py", line 41, in main
    return extension.main(args=args)
  File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 81, in main
    return main(args)
  File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 98, in main
    subscriber(
  File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/ros2topic/verb/echo.py", line 110, in subscriber
    node.create_subscription(
  File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/rclpy/node.py", line 1201, in create_subscription
    check_for_type_support(msg_type)
  File "/Users/snehaldb/ros2_foxy/ros2-osx/lib/python3.8/site-packages/rclpy/type_support.py", line 20, in check_for_type_support
    ts = msg_type.__class__._TYPE_SUPPORT
AttributeError: type object 'NoneType' has no attribute '_TYPE_SUPPORT' This might be a ROS 1 message type but it should be a ROS 2 message type. Make sure to source your ROS 2 workspace after your ROS 1 workspace.

#### Additional information

Works fine if the topic exists.

clalancette, 2020-05-28:
Confirmed; I also get the same behavior on Linux.

jacobperron, 2020-05-28:
A git bisect shows that the bug was introduced in https://github.com/ros2/ros2cli/pull/494

jacobperron, 2020-05-28:
Proposed fix in https://github.com/ros2/ros2cli/pull/520

@snehaldb90 Thanks for the bug report!

PR #520 (2020-05-28): Guard against passing None to rclpy subscriber
Fixes #519

---

I'd like to add a regression test before merging.
commit b08b6b05 (2020-05-28): Guard against passing None to rclpy subscriber (#520)

* Guard against passing None to rclpy subscriber

Fixes #519

* Add regression test
closed 2020-05-28 by jacobperron