ros2/ros2cli #552: 'test_topic_pub_once' test failing

jacobperron, 2020-07-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04 and macOS (Mojave)
- Installation type:
  - source
- Version or commit hash:
  - 95bfb056df795b6eef8bb73b165db31fcf2d867a

#### Steps to reproduce issue

See failing CI jobs, for example:

- https://ci.ros2.org/job/nightly_linux_repeated/1925/testReport/ros2topic.ros2topic.test/test_cli/test_cli/
- https://ci.ros2.org/job/nightly_linux-aarch64_repeated/1248/testReport/ros2topic.ros2topic.test/test_cli/test_cli/
- https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2019/testReport/ros2topic.ros2topic.test/test_cli/test_cli/

#### Expected behavior

Tests pass

#### Actual behavior

Tests sometimes failing

#### Additional information

The tests are flaky; they do not occur every night. When they fail, it is not tied to one particular RMW implementation.

cottsay, 2020-08-05:
This is back already, so clearly my change didn't fix it: http://build.ros2.org/job/Fci__nightly-debug_ubuntu_focal_amd64/170/testReport/junit/ros2topic.test/test_cli/test_cli/

jacobperron, 2020-11-18:
I think this was actually fixed by the referenced PR (for master).

The Foxy CI failure should have been fixed the backport of this in https://github.com/ros2/ros2cli/pull/565.

PR #555 (2020-07-31): Use reliable QoS for ros2topic tests
The `TestROS2TopicCLI` tests perform feature testing of the ros2topic command line interface. If the system is under stress during these tests, messages may be lost (by design). If that happens, there is a fairly high likelihood that the `test_topic_pub_once` test will fail because there is only one opportunity for the message to be successfully transported. We're likely dropping other messages in this suite, but the other tests continuously publish until one of the messages is received (or a timeout occurs), making them significantly less likely to fail.

 [... cut]
closed 2020-11-18 by jacobperron