ros2/ros2cli #560: rename 'ros2' command to 'ros'

christianrauch, 2020-08-04:
## Feature request

Consider dropping the `2` from ROS command-line executables and rename the `ros2` to simply `ros` for a simpler CLI syntax.

#### Feature description
The ROS2 command-line tool unnecessarily carries the version number `2`. The version number can be dropped as there are no other conflicting `ros` executables in ROS1.

E.g.:
`ros2 topic list` ⇨ `ros topic list`
`ros2 run` ⇨ `ros run`

#### Implementation considerations
An implementation only needs to consider renaming all references to `ros2` scripts. The most amount of work has to be done for updating the documentation about the ROS2 CLI.

clalancette, 2020-08-04:
Can you explain more what the perceived benefits of a rename are?  To me, this seems like it would result in a lot of work, possible confusion with the ROS 1 command-line tools (which are still supported), and a long period of transition.  So there would need to be a really strong benefit to convince me.

christianrauch, 2020-08-05:
> Can you explain more what the perceived benefits of a rename are?

This is maybe just a personal preference, but a simple `ros` command is just shorter and more memorable than `ros2`, and it is also closer to the ROS1 syntax. I think it is a benefit if the command-line syntax is similar, e.g. `rostopic list` (ROS1) and `ros topic list` (ROS2). I don't see the reason why the command-line interface has to be changed that much between major versions. There is a reason why commands in a standard Ubuntu environment keep their names between versions (there is no `apt2`, `htop2`, `cp8` etc.).

> To me, this seems like it would result in a lot of work

Backwards compatibility can always be ensured by a symlink and documentation could be updated step-by-step.

> possible confusion with the ROS 1 command-line tools

I don't see a conflict with ROS1 since there is no command with the same name and sourcing both versions at the same time seems to be unsupported anyway.

jacobperron, 2020-08-05:
>  I don't see the reason why the command-line interface has to be changed that much between major versions

We can have ROS 1 and ROS 2 installed side-by-side, with their respective tools. Therefore, we can run `rostopic list` and `ros2 topic list` in the same shell.

> sourcing both versions at the same time seems to be unsupported anyway

This is supported; this is how [ros1_bridge](https://github.com/ros2/ros1_bridge) works.

christianrauch, 2020-08-07:
> We can have ROS 1 and ROS 2 installed side-by-side, with their respective tools. Therefore, we can run `rostopic list` and `ros2 topic list` in the same shell.
> 
> This is supported; this is how [ros1_bridge](https://github.com/ros2/ros1_bridge) works.

If this is fully supported, why do I get the warning `ROS_DISTRO was set to 'noetic' before. Please make sure that the environment does not mix paths from different distributions.`? This seems confusing if it is expected that both distributions are sourced at the same time. In any case, there are no conflicting executable names. If you really insist on sourcing both workspaces, you will see that RViz2 will fail with a symbol lookup error. I doubt that users want to mix workspaces for anything other than the bridge.

But the `rqt` executable / script is available in both ROS versions. If this is fine for `rqt` to have conflicting names, why can't we have non-conflicting commands `rostopic` and `ros topic`?

What technical limitation speaks against having a `ros` symlink to `ros2`?

clalancette, 2020-08-07:
> I don't see a conflict with ROS1 since there is no command with the same name and sourcing both versions at the same time seems to be unsupported anyway.

There's no direct conflict with ROS 1 tools, but I think when explaining the tools it is easy to forget the space.

You can source both ROS 1 and ROS 2 in the same workspace; that is supported for the purposes of running the bridge.

> What technical limitation speaks against having a `ros` symlink to `ros2`?

There's no technical reason.  But there is a documentation/cultural reason.  I still have to stick by my original opinion that it is too close to the originals (causing possible confusion), and it would generate a lot of work in the documentation.  The minor benefits of it being easier to type don't seem to outweigh that.

sloretz, 2020-08-27:
The benefit of less typing feels smaller than the risk of confusion, but there will be less potential for confusion when there are no active ROS 1 distros.

I would think it would be okay to add a `ros` alias on the first ROS 2 distro that doesn't have any target platforms overlapping with ROS 1, and deprecate `ros2` when the last ROS 1 distro goes EOL.

ivanpauno, 2020-08-27:
It has been discussed in today's triage meeting, as previously commente, renaming `ros2` command as `ros` would case confusion in all platforms where ros2 coexists with ros.

In a future release where `ros2` is the only ros version in the platform, this decision can be reconsidered.

closed 2020-08-27 by ivanpauno