ros2/ros2cli #569: param list -t for consistency

JuliusSustarevas, 2020-10-10:
#96  Feature request
ros2 topic list
and 
ros2 service list
have the -t parameter to show types. But ros2 param list does not. It would be convinient and consistent if it did. 

#### Feature description
"ros2 param list -t" could return param names followed by types

clalancette, 2020-10-11:
> "ros2 param list -t" could return param names followed by types

While I like the idea of the feature, I think we may want to use a different flag than `-t`.  The `-t` flag implicitly means print the ROS *message* type, which is not what you are asking for.  Instead, you are asking for the internal stored type of the parameters.  I'll just throw out `--param-type` as a flag idea for now, but maybe others disagree.

In any case, while I think this is a great idea, I don't think we're going to have time to work on it right now.  But if you'd like to contribute a change to add it, we'd be happy to review it.

Thanks for the report.

JuliusSustarevas, 2020-10-11:
Well I personally always wanted params to be ROS Message type which gives a certain nice consistency. You could set some param as a Pose message, set your deadzone speed as a Twist Message etc. But thats a separate feature request i guess

fujitatomoya, 2020-10-28:
@clalancette 

since this is tagged with `help wanted`, i did come up with quick candidate. could you take a look at https://github.com/ros2/ros2cli/pull/572?

fujitatomoya, 2020-10-30:
@clalancette 

friengly ping, i think we can close this.

clalancette, 2020-10-30:
Closing since #572 is now merged.

closed 2020-10-30 by clalancette