ros2/ros2cli #573: 👩‍🌾 test_with_no_daemon_running failing on Windows Repeat

chapulina, 2020-11-02:
The test failed the past 5/6 builds. The first failure was in 2020-10-28:

https://ci.ros2.org/view/nightly/job/nightly_win_rep/2050/testReport/ros2cli.ros2cli.test/test_strategy/test_with_no_daemon_running_3_3_/history/

Failure log:

```
test\test_strategy.py:51: in test_with_no_daemon_running
    assert node._daemon_node is None
E   assert <ros2cli.node.daemon.DaemonNode object at 0x000002D7E3521D30> is None
E    +  where <ros2cli.node.daemon.DaemonNode object at 0x000002D7E3521D30> = <ros2cli.node.strategy.NodeStrategy object at 0x000002D7E3521B20>._daemon_node
```

clalancette, 2020-11-17:
The last failure of this was in Nov 5: https://ci.ros2.org/view/nightly/job/nightly_win_rep/2053/#showFailuresLink .  Since then, we've had almost 2 weeks worth of tests and no failure.  I don't have a pointer to something that definitively fixed this, but given how often it was failing before, I would say that this is fixed.  So I'm going to close this, but feel free to reopen if it pops back up.

jacobperron, 2020-11-18:
Looks like it's still an issue: https://github.com/ros2/ros2cli/issues/576

mjcarroll, 2021-03-17:
This has been failing for the past 60 builds on nightly_win_rep, I think it's still an issue.

https://ci.ros2.org/job/nightly_win_rep/2184/testReport/ros2cli.ros2cli.test/test_strategy/test_with_no_daemon_running_3_3_/

clalancette, 2021-03-17:
Yeah, it is likely still an issue.  It looks to be platform-independent though (see #576), so I'm going to reclose this one and update the description of #576 instead.

closed 2021-03-17 by clalancette