ros2/ros2cli #574: Could not get publisher / subscriptions info from topic through XMLRPC

cwyark, 2020-11-06:
Hi, I'm trying to get publisher and subscription info by topic through XMLRPC interface, but get unknown tag when invoking the following functions. 

1. get_publishers_info_by_topic(topic_name)
2. get_subscriptions_info_by_topic(topic_name)

## Bug report

**Required Info:**

- Operating System:
  - OSX / Ubuntu 20.04
- Installation type:
  - binary 
- Version or commit hash:
  - FOXY latest release
- DDS implementation:
  -  ALL implementations
- Client library (if applicable):
  - none

#### Steps to reproduce issue

```
cwyark@geek:[ros2_ws]: python3
Python 3.6.8 (default, Sep 30 2019, 22:08:18) 
[GCC 4.2.1 Compatible Apple LLVM 10.0.1 (clang-1001.0.46.4)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import xmlrpc
>>> from xmlrpc import client
>>> p = client.ServerProxy('http://localhost:11523/ros2cli/')
>>> p.get_topic_names_and_types()
[['/chatter', ['std_msgs/msg/String']], ['/parameter_events', ['rcl_interfaces/msg/ParameterEvent']], ['/rosout', ['rcl_interfaces/msg/Log']]]
>>> p.get_publishers_info_by_topic('/chatter')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/cwyark/.pyenv/versions/3.6.8/lib/python3.6/xmlrpc/client.py", line 1112, in __call__
    return self.__send(self.__name, args)
  File "/Users/cwyark/.pyenv/versions/3.6.8/lib/python3.6/xmlrpc/client.py", line 1452, in __request
    verbose=self.__verbose
  File "/Users/cwyark/.pyenv/versions/3.6.8/lib/python3.6/xmlrpc/client.py", line 1154, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/Users/cwyark/.pyenv/versions/3.6.8/lib/python3.6/xmlrpc/client.py", line 1170, in single_request
    return self.parse_response(resp)
  File "/Users/cwyark/.pyenv/versions/3.6.8/lib/python3.6/xmlrpc/client.py", line 1336, in parse_response
    p.feed(data)
  File "/Users/cwyark/.pyenv/versions/3.6.8/lib/python3.6/xmlrpc/client.py", line 439, in feed
    self._parser.Parse(data, 0)
  File "/private/var/folders/pq/_2x0qf_n70xf3mnpp1xdd_s00000gn/T/python-build.20190930220651.51176/Python-3.6.8/Modules/pyexpat.c", line 419, in StartElement
  File "/Users/cwyark/.pyenv/versions/3.6.8/lib/python3.6/xmlrpc/client.py", line 677, in start
    raise ResponseError("unknown tag %r" % tag)
xmlrpc.client.ResponseError: ResponseError("unknown tag 'rclpy.topic_endpoint_info.TopicEndpointInfo'",)
```

#### Expected behavior

We could use any XMLRPC tool to get the publisher info by topic name. 

#### Actual behavior

Got exception when invoking get_publishers_info_by_topic and get_subscriptions_info_by_topics

#### Additional information

I tried use python rclpy node to get these information without any problems, so I guess the problem happened at serialize / deserialize XMLRPC message at server side. 
----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2020-11-06:
Yeah, it looks like the daemon is currently relying on the client to provide `TopicEndpointInfo` somehow.  I'm not entirely sure why.

This isn't likely something we'll look at soon.  But please feel free to open a pull request if you find the issue.

cwyark, 2020-11-07:
Looks like the tag name is out of scope from xmlrpc generic type. e.g. <rclpy.topic_endpoint_info.TopicEndpointInfo>. Most xmlrpc client might not know what it is. (http://xmlrpc.com/spec.md)
[image]

I think one solution is to write custom handler for these tags in their xmlrpc client tool.

fujitatomoya, 2020-11-08:
`get_publishers_info_by_topic` and `get_subscriptions_info_by_topic` returns List[TopicEndpointInfo], so general xmlrpc client cannot handle `TopicEndpointInfo` since it is unaware of. I do not think custom handler would be a good idea since burden goes to user. instead of doing that, maybe we could replace `TopicEndpointInfo` into `List` and `Tuple` with generic types.

cwyark, 2020-11-09:
Agree with that, maybe we can change ``rclpy.topic_endpoint_info.TopicEndpointInfo`` and ``rclpy.topic_endpoint_info.TopicEndpointTypeEnum`` to other generic type. 

## rclpy.topic_endpoint_info.TopicEndpointInfo 

I look into the structure from wireshark.
[image]
In a short term, it will be like this, 
```
<array>
    <data>
        <rclpy.topic_endpoint_info.TopicEndpointInfo>
            <struct>
                <member/>
                <member/>
                <member/>
                <member/>
            </struct>
        </rclpy.topic_endpoint_info.TopicEndpointInfo>
        <rclpy.topic_endpoint_info.TopicEndpointInfo>
            <struct>
                <member/>
                <member/>
                <member/>
                <member/>
            </struct>
        </rclpy.topic_endpoint_info.TopicEndpointInfo>
    </data>
</array>
```

I think we can just remove that unknown tag (rclpy.topic_endpoint_info.TopicEndpointInfo). like this: 

```
<array>
    <data>
            <struct>
                <member/>
                <member/>
                <member/>
                <member/>
            </struct>
            <struct>
                <member/>
                <member/>
                <member/>
                <member/>
            </struct>
    </data>
</array>
```
This approach still keeps xmlrpc specification and acceptable with most xmlrpc client. 

## rclpy.topic_endpoint_info.TopicEndpointTypeEnum

[image]

```
<value>
    <int>
        1
    </int>
</value>
```

## rclpy.qos.QosProfile

[image]

rclpy.qos.QosProfile is like rclpy.topic_endpoint_info.TopicEndpointInfo , and remove that tag. just replace with struct tag.

gavanderhoorn, 2024-12-11:
It's not nice, but until the XMLRPC server marshals the `rclpy` types using standard XMLRPC types, `from ros2cli.xmlrpc import marshal` registers the custom (de)serialisers and allows standard `xmlrpc.client.ServerProxy` instances to interact with the daemon's server.