ros2/ros2cli #576: 👨‍🌾 test_with_no_daemon_running is flaky on nightly OSX repeated.

azeey, 2020-11-17:
Failing builds:
- 2020-11-15: https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2158/testReport/junit/ros2cli.ros2cli.test/test_strategy/test_with_no_daemon_running_3_3_/
- 2020-10-29: https://ci.ros2.org/view/nightly/job/nightly_osx_repeated/2140/testReport/junit/ros2cli.ros2cli.test/test_strategy/test_with_no_daemon_running_3_3_/

Stacktrace:

```
test/test_strategy.py:51: in test_with_no_daemon_running
    assert node._daemon_node is None
E   assert <ros2cli.node.daemon.DaemonNode object at 0x115667fd0> is None
E    +  where <ros2cli.node.daemon.DaemonNode object at 0x115667fd0> = <ros2cli.node.strategy.NodeStrategy object at 0x1156670a0>._daemon_node
```

jacobperron, 2020-11-18:
Same as https://github.com/ros2/ros2cli/issues/573, but on OSX. It appears platform independent.

clalancette, 2021-03-17:
Also failing on Windows repeated: https://ci.ros2.org/job/nightly_win_rep/2184/testReport/ros2cli.ros2cli.test/test_strategy/test_with_no_daemon_running_3_3_/

clalancette, 2022-04-08:
I went back looking through the Windows repeated jobs for the last month, and I saw no evidence of this failed test in any of those.  I don't have a PR that fixed it, nor do I have direct evidence that this was fixed, but I'm going to close this out for now.  If it comes back, we can open a new issue about it.

closed 2022-04-08 by clalancette