ros2/ros2cli #577: test_echo_pub.TestROS2TopicEchoPub failing (Rdev and RPr)

jacobperron, 2020-11-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.10.0
- DDS implementation:
  - Fast-DDS

#### Steps to reproduce issue

I haven't been able to reproduce any of the test failures locally.

#### Expected behavior

All tests pass.

#### Actual behavior

test_echo_pub.TestROS2TopicEchoPub is failing in the latest [dev job](http://build.ros2.org/view/Rdev/job/Rdev__ros2cli__ubuntu_focal_amd64/21) and RPr jobs ([for example](http://build.ros2.org/job/Rpr__ros2cli__ubuntu_focal_amd64/121/testReport/ros2topic.ros2topic.test/test_echo_pub/test_echo_pub)).

#### Additional information

I've seen [test_api.TestROS2DoctorAPI](http://build.ros2.org/job/Rpr__ros2cli__ubuntu_focal_amd64/121/testReport/ros2doctor.ros2doctor.test/test_api/test_api/) also fail sometimes during a RPr job. Seems to also be related to an issue connecting with a test publisher.

fujitatomoya, 2020-11-24:
I cannot reproduce this issue locally either. So far I am not sure what is wrong with test code, any clues? maybe something we can do is to wait for number of publication or subscription before CLI is issued?

jacobperron, 2020-12-21:
The past couple dev jobs haven't showed this error, but it looks like a recent PR job failure could be related: https://build.ros2.org/job/Rpr__ros2cli__ubuntu_focal_amd64/128/testReport/

Looks like a rare flake.

clalancette, 2022-02-03:
This doesn't show up in any of the recent Rdev jobs.  I'm going to close this out, but we can reopen if this appears again.

closed 2022-02-03 by clalancette